hideakitai/ArtNet #3: Compile error

Cubyte, 2019-08-02:
Not sure but i get this error when i compiled the example
'ArtnetReceiver' does not name a type
what am i doing wrong

hideakitai, 2019-10-11:
Sorry for late reply, which Arduino board do you use?

hideakitai, 2020-06-27:
@Cubyte I've supported more boards on v0.1.1, check it out please!

closed 2020-06-27 by hideakitai

hideakitai, 2022-03-23:
@conraddang Hi, maybe your ide configuration is not working. First, please use library manager if you are not familiar with Arduino. If you still have the problem, please open a new issue with the following information.

- IDE (Arduino IDE, PlatformIO, etc.) : Arduino IDE ?
- Board/MCU: ESP32-WROOM32 ?
- Board Firmware Version: v2.0.2 ?
- Library Version: ArtNet v0.2.12 ?
- Reproducable & Minimal & Compilable Code
- Additional Information (System chart, etc.)

conraddang, 2022-03-23:
Hi, thanks a lot! I really appreciate your help. I will try that out. If I still encounter the problem afterwards, i will open a new issue with the details that you need.