hideakitai/ArtNet #4: Arduino DUE

CurtLemon, 2020-06-23:
Is it possible to use this library with the ARM chip on the Arduino Due?

hideakitai, 2020-06-25:
@CurtLemon Sorry for the late reply, currently no. But I will try it in a few days!

hideakitai, 2020-06-27:
@CurtLemon Supported on v0.1.1. If you have any problem, please feel free to let me know!

closed 2020-06-27 by hideakitai

CurtLemon, 2020-06-28:
Thank you so much!  Works great, this is the only library I’ve found that can both send and receive Artnet, and handle it in a clean way. I really appreciate your making this library available and getting back to me so quickly. Thanks Tim Boland From: Hideaki TaiSent: Saturday, June 27, 2020 3:23 AMTo: hideakitai/ArtNetCc: CurtLemon; MentionSubject: Re: [hideakitai/ArtNet] Arduino DUE (#4) Closed #4.