hideakitai/ArtNet #5: Receiver unsubscribe function

topjor, 2020-09-22:
I would like to have a function to cancel / remove / unsubscribe callback functions, in my case for a given universe.

I will be looking into making a PR myself but that probably will be somewhere this or next week.

hideakitai, 2020-09-23:
Hi, thanks.
I also try to add them if I have time :)

hideakitai, 2020-09-24:
May be it works, if you have a problem, let me know and reopen this issue. Thanks!

closed 2020-09-24 by hideakitai

topjor, 2020-09-24:
Thank you,
I would suggest to unsubscribe from all callbacks when no universe is given as someone might be subscribed to a specific universe or multiple and wants to clear all.

hideakitai, 2020-09-30:
`subscribe(func)` and `unsubscribe()` is for the callback of every packet in every universe, so I added `clear_subscribers()` to unsubscribe all callbacks.