hideakitai/ArtNet #6: Error compiling receiving example

pieterVE, 2020-10-10:
I just started experimenting with an Arduino Ethernet board to make a Artnet to DMX converter.
When compiling the receiving example I bumped into following error. I just installed the Arduino compiler and enabled the Artnet library.... Is there something more I should do?

Arduino: 1.8.13 (Mac OS X), Board:"Arduino Ethernet"

In file included from ../Documents/Arduino/libraries/ArtNet/util/ArxTypeTraits/ArxTypeTraits/type_traits.h:407:0,
                 from ../Documents/Arduino/libraries/ArtNet/util/ArxTypeTraits/ArxTypeTraits.h:40,
                 from ../Documents/Arduino/libraries/ArtNet/Artnet.h:10,
                 from ../Documents/Arduino/my_receiver/my_receiver.ino:1:
../Documents/Arduino/libraries/ArtNet/util/ArxTypeTraits/ArxTypeTraits/functional.h: In instantiation of 'arx::stdx::function<R(Args ...)>::function(const Func&) [with Func = void(unsigned char*, unsigned int); dF = void (*)(unsigned char*, unsigned int); typename arx::stdx::enable_if<(! arx::stdx::is_same<dF, arx::stdx::function<R(Args ...)> >{})>::type* <anonymous> = 0; typename arx::stdx::enable_if<arx::stdx::is_convertible<typename arx::stdx::details::result_of<dF&(Args ...), void>::type, R>{}>::type* <anonymous> = 0; R = void; Args = {unsigned char*, unsigned int}]':
../Documents/Arduino/my_receiver/my_receiver.ino:32:41:   required from here
../Documents/Arduino/libraries/ArtNet/util/ArxTypeTraits/ArxTypeTraits/functional.h:87:44: error: new cannot be applied to a function type
             data = reinterpret_cast<void*>(new Func(f));
                                            ^~~~~~~~~~~
exit status 1
Fout bij het compileren voor board Arduino Ethernet

hideakitai, 2020-10-11:
@pieterVE Thanks! Maybe current master branch works, could you check if it works?

pieterVE, 2020-10-11:
Thanks for the quick reply. I checked the master branch and that is working fine!  I was able to receive the artnet data on my Arduino board...

hideakitai, 2020-10-11:
Glad to hear that:)
Please let me know if you have any problem.

closed 2020-10-11 by hideakitai