hideakitai/ArtNet #7: How would you go about connecting the boards?

jackcrane, 2020-10-13:
Hi!

Thanks for your great work, this library looks really easy to use. How do you connect the boards to start with? Do we need to init a wifi direct relationship?

Thanks

hideakitai, 2020-10-14:
Hi, there are many ArtNet controllable devices, but if you want to use ArtNet over WiFi, you can use wireless ArtNet device like this: 
[AIR2DMX PRO](https://www.amazon.co.uk/AIR2DMX-WLAN-Interface-ArtNet-Node/dp/B07Z8FWJQG)

Of course you can create such a wireless ArtNet-Dmx controller by yourself:
[ESP8266 Artnet to DMX](https://www.instructables.com/ESP8266-Artnet-to-DMX/)

Or more handy one is using ESP32 and NeoPixel (the library used in this article is not mine, thogh :) ):
[Using Artnet DMX and the ESP32 to Drive Pixels](https://learn.sparkfun.com/tutorials/using-artnet-dmx-and-the-esp32-to-drive-pixels/all)

hideakitai, 2020-10-14:
If you need the example to use ArtNet and NeoPixel (FastLED), I can create it. Let me know if you want :)

hideakitai, 2020-10-19:
Ok, please re-open this issue if you have any questions!

closed 2020-10-19 by hideakitai