hideakitai/ArtNet #8: The sender example doesn't work

blushift80, 2020-10-19:
Hi, congratulations for the excellent work!
I tried running the sender.ino sample sketch in my Arduino Uno board with Ethernet Shield (W5100). Sketch freezes when streaming. The ip address ping is not responding and no udp (artnet) packet is sent to my target device.
Doing some tests I realized that the "packet.size()" function in the send() function in Artnet.h returns 0. Is this normal?
Is the project not compatible with my hardware?
Thank you.

hideakitai, 2020-10-20:
Thanks for reporting.
I've fixed the issue and confirmed it works.

- New Arduino Uno R3 (light green) + Ethernet Shiled 2 (W5500)
- Arduino 1.8.13 + AVR 1.8.3

But I've encountered the another issue discussed here:
https://github.com/arduino-libraries/Ethernet/issues/78

The configuration is:
- Old Arduino Uno R3 (deep blue) + Ethernet Shiled 2 (W5500)
- Arduino 1.8.13 + AVR 1.8.3

After replacing Uno to another one, it works fine, so please note about the configuration.
My old Arduino Uno R3 may have been broken, though :-(

If you still have any problem, let me know. 
Thanks, again!

closed 2020-10-20 by hideakitai

njrl, 2020-11-06:
Hi,
I've a similar problem  ... the the artnet sender does not send ... the whole (sender example) sketch  runs a bit sluggish but it runs ... but no packets are delivered ... tested with wireshark
IDE: 1.8.13 linux
library: 0.1.9
arduino: Mega
ethernet: W5100
any hints for debugging ?

regards

hideakitai, 2020-11-06:
Hi, sorry I don't have Mega and W5100 Ethernet Shield, so I can't test it... But as mentioned above, I've confirmed this library woks on Uno and Ethernet Shield2. Mega is almost same and more powerful than Uno, so probably this library works also on Mega.

Can you check if this library works on same configuration?
In addition, please check again about:
- ping command (IP address and subnet)
- universe setting

I also added the application example which works in my environment. (If hope you can use or try to use TouchDesigner...!)
https://github.com/hideakitai/ArtNet/tree/master/examples/Apps/TouchDesigner