hideakitai/ArtNet #10: How do i send artnet command to a specific channel?

silentonset, 2020-11-08:
Hi,

I am new to arduino and the ethershield, i find your library very easy to understand unfortunately i can't find documentation on how to send artnet to a specifc address. Play around your library, i can only send to a group of channels via 
[ const uint16_t size = 512; ]

Can you also explain what "data[size]" and "memset(data, value, size);" does?

Thanks!

hideakitai, 2020-11-08:
Hi, I've transferred this issue to ArtNet. (Right?)

ArtNet data can be up to 512 bytes of data. So I've made an array for 512 bytes as `data[size]`.
And `memset` sets the `value` from `data[0]` to `data[511]`. 
I've also added the example to use NeoPixel with ArtNet.

For more details, please refer [spec](https://artisticlicence.com/WebSiteMaster/User%20Guides/art-net.pdf) and [packet definition](https://art-net.org.uk/structure/streaming-packets/artdmx-packet-definition/).

closed 2020-11-08 by hideakitai

silentonset, 2020-11-09:
Thanks hideakitai ! All is good.