hideakitai/ArtNet #11: Send & Receive not working

silentonset, 2020-11-09:
Hi hideakitai,

I've been doing some test with the send_receive and it doesn't seem to be working as expected.

When i use Receive Sketch:
I get :
artnet data (universe : 1) = 
215,149,62,159,64,0,0,0,0,0,0,0,0,0,0... etc_

But when i use send_receive
i get :
_set subscriber
start_

It doesn't seem to execute the function, is the implementation different from the Receive sketch?

hideakitai, 2020-11-09:
Hi, send_receive's receiver is completely same as receiver example. Can you check ping command from your pc to your device first, and then please check the universe setting.

If everything is fine, let me know your IDE version, board name, board version, ArtNet library version, and sketch if you can.

silentonset, 2020-11-09:
Hi hideakitai,

All my setting are exactly the same. in fact when i trigger artnet channel 1 @ universe 1.

the serial monitor respond with "set subscriber start" everytime i trigger it

[image]

i am using an UNO R3 Board (BN: Arduino Uno VID: 2341PID: 0043 )
with a Hanrun HR911105A (19/06) ethershield (W5100).
i am using Arduino 1.8.13 of the IDE and the latest Artnet Library  v0.1.10

When i compile, i have an error that says "Not enough memory" so i edit the sketch 
"const uint16_t size = 512;" to "const uint16_t size = 10;"

this is the log i get :

Sketch uses 13532 bytes (41%) of program storage space. Maximum is 32256 bytes.
Global variables use 1745 bytes (85%) of dynamic memory, leaving 303 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
Invalid library found in /Users/mbpr-2013/Documents/Arduino/libraries/hardware: no headers files (.h) found in /Users/mbpr-2013/Documents/Arduino/libraries/hardware
Invalid library found in /Users/mbpr-2013/Documents/Arduino/libraries/DMX_Shield: no headers files (.h) found in /Users/mbpr-2013/Documents/Arduino/libraries/DMX_Shield

hideakitai, 2020-11-10:
@silentonset Integrated sender/receiver is not available on Arduino UNO. Because it doesn't have enough memory, it hangs when both sender and receiver instance are used as noted in README.

> Some boards without enough memory may not be able to use integrated sender/receiver. Please use only sender OR receiver.

Please consider to use more powerful boards or just use one of sender or receiver. I don't think we're going to use both of them in the same board very often.

hideakitai, 2020-11-10:
If you have any other question, let me know and reopen issue. Thanks.

closed 2020-11-10 by hideakitai

silentonset, 2020-11-11:
Will do, i am trying to get a new board. Will revert once i test on the new board.

hideakitai, 2020-11-11:
Ok, I hope you can make it!

silentonset, 2020-11-11:
Do you have any recommended boards?

hideakitai, 2020-11-11:
It's depending on the situation. What kind of functions do you need? Sending artnet, receiving artnet and what else?

silentonset, 2020-11-11:
I am hoping to add a dmx shield as well, i am looking to build a dmx/artnet/sACN e131 receiver to trigger a few contact relay and also have to detect button trigger to trigger artnet/dmx/sACN e131.

hideakitai, 2020-11-11:
To use shields for Uno, Arduino Mega is reasonable and more powerful than Uno. Good luck.

silentonset, 2020-11-12:
Hi hideaki, i bought an arduino due and it works great!!! Thanks!

hideakitai, 2020-11-12:
Great to fear that! Hope you can make it :-)