hideakitai/ArtNet #12: How to use broadcast?

omgkenny6135, 2020-12-09:
Hi, I'm want to use this library to connect with Qlab, but Qlab only can find the Art-Net drive when the Art-Net drive have broadcast.
So how do I fix it to broadcast on network?

hideakitai, 2020-12-21:
Hi, you can set any address for `artnet.begin(your_ip_addr)`. Please set your own broadcast address here e.g. `artnet.begin("192.168.0.255")`.

omgkenny6135, 2020-12-22:
Hi Hideakitai,
Thanks for your reply,
However, whether I replace artnet.begin("192.168.0.255") with artnet.begin() or add artnet.begin("192.168.0.255") in a new line, the result is the same: the LED light does not work but it can be used Command Prompt to Ping to Arduino IP;
And all Art-Net software still cannot find my Arduino device through broadcast.

I'm using Mega2560 Board, and use sk6812 LED strip.

omgkenny6135, 2020-12-22:
> 嗨，您可以為設置任何地址`artnet.begin(your_ip_addr)`。請在此處設置您自己的廣播地址，例如`artnet.begin("192.168.0.255")`。如果仍然有問題，可以重新打開問題。

Hi Hideakitai,
Thanks for your reply,
However, whether I replace artnet.begin("192.168.0.255") with artnet.begin() or add artnet.begin("192.168.0.255") in a new line, the result is the same: the LED light does not work but it can be used Command Prompt to Ping to Arduino IP;
And all Art-Net software still cannot find my Arduino device through broadcast.

I'm using Mega2560 Board, and use sk6812 LED strip.

omgkenny6135, 2020-12-22:
> Hi, you can set any address for `artnet.begin(your_ip_addr)`. Please set your own broadcast address here e.g. `artnet.begin("192.168.0.255")`. If you still have problem, you can reopen the issue.

Hi Hideakitai,
Thanks for your reply,
However, whether I replace artnet.begin("192.168.0.255") with artnet.begin() or add artnet.begin("192.168.0.255") in a new line, the result is the same: the LED light does not work but it can be used Command Prompt to Ping to Arduino IP;
And all Art-Net software still cannot find my Arduino device through broadcast.

I'm using Mega2560 Board, and use sk6812 LED strip.

hideakitai, 2020-12-23:
Ok, please let me know the development environment (versions, IDEs, etc.), simplified system diagram (network sturcture and IPs), and so on.

omgkenny6135, 2020-12-26:
> Ok, please let me know the development environment (versions, IDEs, etc.), simplified system diagram (network sturcture and IPs), and so on.

My Arduino IDE versions is 1.8.13, My code is as follows:

`#include <Adafruit_NeoPixel.h>
#include <Artnet.h>

const int numLeds = 100;
const byte dataPin = 9;
Adafruit_NeoPixel leds = Adafruit_NeoPixel(numLeds, dataPin, NEO_GRBW + NEO_KHZ800);

// Ethernet stuff
const IPAddress ip(192, 168, 0, 164);
uint8_t mac[] = {0x67, 0x89, 0xAB, 0x01, 0x23, 0x45};

ArtnetReceiver artnet;
byte NetID = {0};
byte SubnetID = {0};
byte UniverseID = {0};
short universe = (NetID * 256 + SubnetID*16 + UniverseID);

void setup() {
  Ethernet.init(53);
  leds.begin();
  delay(2000);

  Ethernet.begin(mac, ip);
  artnet.begin("192.168.0.255");

  // if Artnet packet comes to this universe, this function is called
  artnet.subscribe(universe, [&](const uint8_t* data, const uint16_t size) {
    for (int i = 0; i < numLeds; ++i) {

      int RDC = data[i * 3];  //RedDataChannel
      int GDC = data[i * 3 + 1];  //GreenDataChannel
      int BDC = data[i * 3 + 2];  //BlueDataChannel

        leds.setPixelColor(i, RDC, GDC, BDC, 0);
    }
    leds.show();
  });

}

void loop() {
  artnet.parse();  // check if artnet packet has come and execute callback
}`

Ethernet device configuration diagram:
PC (IP : 192.168.0.152, Art-Net software: touchdesigner 2020.27390, jinx-2.4)
│
Network switch (IP : 192.168.0.1 , Subnet : 255.255.255.0)
│
w5500 (IP : 192.168.0.164) --- Arduino Mega 2560 PRO (Embed)

hideakitai, 2020-12-26:
Thanks. I also want to know:
- Mega's version (avr)
- ArtNet library's version
- Universe settings
- Which is the sender and which is the receiver?
- The sending/receiving works correctly without broadcast?

omgkenny6135, 2020-12-26:
> Thanks. I also want to know:
> 
> * Mega's version (avr)
> * ArtNet library's version
> * Universe settings
> * Which is the sender and which is the receiver?
> * The sending/receiving works correctly without broadcast?

Mega's version is ATMEGA 2560
ArtNet library's version is v0.1.11
Universe settings : `byte NetID = {0};
byte SubnetID = {0};
byte UniverseID = {0};
short universe = (NetID * 256 + SubnetID*16 + UniverseID);`
PC is the Art-Net sender, and Arduino is the receiver.
If I don't change `artnet.begin()` to `artnet.begin("192.168.0.255")` then Arduino receiving well work, but if I change it, then Art-Net will not work, but stell can use Command Prompt to Ping to Arduino

hideakitai, 2020-12-27:
Thanks. The AVR board firmware version is the latest? (v1.8.3)

If you are sending from TouchDesigner, Arduino should have fixed IP. `artnet.begin(destination_ip)` sets the destination, so it should be blank if you just receive the artnet packet.

What do you mean "broadcast"? Where do you want to broadcast ArtNet packet from and to??

omgkenny6135, 2020-12-27:
> 謝謝。AVR板固件版本是最新的嗎？（v1.8.3）
> 
> 如果從TouchDesigner發送，則Arduino應該具有固定的IP。`artnet.begin(destination_ip)`設置目的地，因此如果您僅收到artnet數據包，則應為空白。Arduino本身的IP由來配置`Ethernet.begin()`。這樣您就可以收到ping了。
> 
> 您是什麼意思“廣播”？您想在哪裡廣播ArtNet數據包？

Can you tell me 

> Thanks. The AVR board firmware version is the latest? (v1.8.3)
> 
> If you are sending from TouchDesigner, Arduino should have fixed IP. `artnet.begin(destination_ip)` sets the destination, so it should be blank if you just receive the artnet packet. And the IP of Arduino itself is configured by `Ethernet.begin()`. So you can receive ping.
> 
> What do you mean "broadcast"? Where do you want to broadcast ArtNet packet from and to??

Can you tell me how to get AVR board firmware version?

If I use TouchDesigner or Jinx(http://www.live-leds.de/downloads/), it works very good.
But if I want to use QLab to control my Arduino Art-Net drive, then I need to broadcast Art-Net on local area network.

I think it called "Directed Broadcast".

Spec (Art-Net 4) page 7:
Directed Broadcast: When a network first connects, the Controller does not know the
number of Nodes on the network, nor does it know their IP addresses. The Directed
broadcast address allows the Controller to send an ArtPoll to all Nodes on the network.

hideakitai, 2020-12-27:
@omgkenny6135 Ok, I've got what you want to, and supported it maybe. Can you check if the latest master branch works?

omgkenny6135, 2020-12-28:
@hideakitai I try to cheage `ArtnetReceiver` to `Artnet artnet` but the light didn't works.
Also I don't know how to transfer ArtPoll and ArtPollReply, Is it the same to use `artnet.begin("192.168.0.255")` to transfer it?

hideakitai, 2020-12-28:
@omgkenny6135 ArtPoll and ArtPollReply works automatically if the implementation works correctly. Just use following environment. No need to set broadcasts.

> If I use TouchDesigner or Jinx(http://www.live-leds.de/downloads/), it works very good.

omgkenny6135, 2020-12-31:
Hi hideakitai, in the past few days I tried to change my code, but those code didn't works.

I uploaded my last program that worked normally to my Github:
https://github.com/omgkenny6135/Fantasy-Pixel-Art-Net-Code/blob/main/artnet_RGB_receive_T5.ino

> @omgkenny6135 ArtPoll and ArtPollReply works automatically if the implementation works correctly. Just use following environment. No need to set broadcasts.
> 
> > If I use TouchDesigner or Jinx(http://www.live-leds.de/downloads/), it works very good.

When there is no ArtPoll and ArtPollReply, if you use the specified IP, TouchDesigner or Jinx can work normally, but because you need to connect with Qlab, I still need ArtPoll and ArtPollReply.

hideakitai, 2021-01-01:
@omgkenny6135 Hi, I've fixed ArtPollReply (though it should be refactored). Can you check if the latest master branch works in your environment?

hideakitai, 2021-01-02:
@omgkenny6135 Hi, I've refactored and maybe latest master branch works automatically with ArtPoll and ArtPollReply. Please use `receiver` or `send_receive` example as is (but you need to change network settings). If Art-Net controller polls devices, this library automatically reply it in the background.

omgkenny6135, 2021-01-02:
> @omgkenny6135 Hi, I've refactored and maybe latest master branch works automatically with ArtPoll and ArtPollReply. Please use `receiver` or `send_receive` example as is (but you need to change [...]

Hi hHideakitai,
I use my old receiver code and it works!
Thanks for your help, and I will test Qlab at next week with my friend, if we don't have any problem, I will close this issue.

hideakitai, 2021-01-03:
Ok, please let me know if you have problems :)

hideakitai, 2021-01-12:
Have you tried to use this library with QLab? I'm going to close this issue but if you have any problem, please reopen the issue.

closed 2021-01-12 by hideakitai

einlichtvogel, 2025-06-01:
Hey @hideakitai,

I currently also try to implement the setArtPollReplyConfig function, but I don't get it working.

This is the code in my setup() function:
`  artnet.setArtPollReplyConfig(
    0x00FF,
    0x0000,
    0x00,
    0x08,
    String("Arduino ArtNet"),
    String("Arduino ArtNet Protocol by hideakitai/ArtNet"),
    String(""),
    (uint8_t[]){0, 0, 0, 0}
  );
artnet.begin();
`

I am using Artnet artnet; as global var.

Can you help me?
I am building a receiving node.

hideakitai, 2025-06-01:
@einlichtvogel  As the code below shows, 0x2000 is Poll and 0x2100 is PollReply. 

https://github.com/hideakitai/ArtNet/blob/1c149e490b3d347831171a402d3a664ce57e40f8/Artnet/Common.h#L15-L17

This library expects to receive a Poll from the controller and replies with a PollReply. So, receiving Poll"Reply" is not supported. 

https://github.com/hideakitai/ArtNet/blob/1c149e490b3d347831171a402d3a664ce57e40f8/Artnet/Receiver.h#L105-L109

Please try to check the real packet when "Unsupported OpCode： 2100" occurs. Also, please open the new issue instead of continuing to discuss in the closed issue.