hideakitai/ArtNet #13: Receiving on ESP32 stops working

cerian, 2020-12-22:
Hi,

Great work on the library! 

The receive example works as expected on an ESP8266 with MadMapper sending ArtNet data to the board. However, using the same example on an ESP32 does not work as expected and it appears as if the ESP32 stops listening for ArtNet messages. The result varies between the ESP32 not receiving any messages to the board receiving a couple of messages (between 2 to 7) when looking at the serial monitor on the receiver example. The ESP32 does not crash and the loop appears to be running as it should.

I have tried five different ESP32 boards with the same result:

- ESP32-DevKitC (from Espressif)
- ESP32-PICO-KIT
- Lolin32 Lite
- ESPDUINO-32
- Olimex ESP32-PoE-ISO

cerian, 2020-12-22:
Looks like it was previously working in v. 0.1.8. So v. 0.1.9 must have broken it somehow

cerian, 2020-12-22:
Reverting this line https://github.com/hideakitai/ArtNet/compare/0.1.8...0.1.9#diff-d6353df15ea32267b7784451307e787420a919c32c3017d9a7fad58e0884afb3R287

Seems to make it work again (also works on newest commit)

hideakitai, 2020-12-23:
@cerian Hi, thank you for reporting. Maybe some packet comes without data, and it remains buffer I think. Can you check if it works with changing this line:

```C++
if (size <= HEADER_SIZE) return false;
```

to 

```C++
if (size <= HEADER_SIZE) {
    uint8_t d[size];
    stream->read(d, size); // discard
    return false;
}
```

hideakitai, 2020-12-23:
In addition, can you print the 'size' of data in the callback with v0.1.8? I think some packet show the strange data size, e.g. size = 0.

cerian, 2020-12-23:
> @cerian Hi, thank you for reporting. Maybe some packet comes without data, and it remains buffer I think. Can you check if it works with changing this line:
>
> ```c++
> if (size <= HEADER_SIZE) return [...]
Confirmed. It works!

> In addition, can you print the 'size' of data in the callback with v0.1.8? I think some packet show the strange data size, e.g. size >= 0.
> 
You mean the uint16_t size parameter in the callback, right? 
For v. 0.1.8 the size is 530. For v. 0.1.9 the size is 512. Both are consistent.

Happy holidays!

hideakitai, 2020-12-24:
@cerian Thank you for testing! I'll fix it in next release.

hideakitai, 2020-12-24:
Fixed!

closed 2020-12-24 by hideakitai