hideakitai/ArtNet #14: Wifi Not Work!

omgkenny6135, 2021-01-09:
Hi Hideakitai, I have a new issue:

I try to use Wifi -> Receiver to work on my nodeMCU, but the error like this:
`'ArtnetWiFiReceiver' does not name a type; did you mean 'ArtnetReceiver'?`

hideakitai, 2021-01-09:
@omgkenny6135 Fixed!

omgkenny6135, 2021-01-09:
> @omgkenny6135 Fixed!

I'm useing examples -> WiFi -> receiver_fastled, but still have same problem.

hideakitai, 2021-01-09:
@omgkenny6135 Have you updated the library to latest master branch (v0.2.1)? I don't have any compile error.

omgkenny6135, 2021-01-09:
> @omgkenny6135 Have you updated the library to latest master branch (v0.2.1)? I don't have any compile error.

Yes, I test the code after I upload the library to v0.2.1!

hideakitai, 2021-01-09:
Please let me know your configuration. I don't have any compile error as shown in the screen shot.
[image]

hideakitai, 2021-01-10:
@omgkenny6135 If you still have the problem, please show me your code (if it is not same as the `receive_fastled` example).

omgkenny6135, 2021-01-10:
> @omgkenny6135 If you still have the problem, please show me your code (if it is not same as the `receive_fastled` example).

I found something cool! If I remove `artnet.forward(universe, leds, NUM_LEDS);` in `receive_fastled`, then it works! But if I don't remove it, then the error will like this:[https://github.com/omgkenny6135/Fantasy-Pixel-Art-Net-Code/blob/main/wifi_error_1](url)

hideakitai, 2021-01-10:
@omgkenny6135 Are you using Ethernet on nodemcu? Which do you want to use for ArtNet, Ethernet or WiFi? I've dropped support for Ethernet on ESP8266 in v0.2.1.

omgkenny6135, 2021-01-10:
> @omgkenny6135 Are you using Ethernet on nodemcu? Which do you want to use for ArtNet, Ethernet or WiFi? I've dropped support for Ethernet on ESP8266 in v0.2.1.

Realy? I'm using v0.2.1 to upload the code to my nodemcu, and using the wifi to connect my Network switch, it works very well!

hideakitai, 2021-01-10:
??? WiFi?? I saw your repository but yoU haven't used WiFi.

Once again, please give me your latest code, I can't debug it. Or I will close this issue because there's no problem in my environment.

omgkenny6135, 2021-01-11:
> ??? WiFi?? I saw your repository but yoU haven't used WiFi.
> 
> Once again, please give me your latest code, I can't debug it. Or I will close this issue because there's no problem in my environment.

I upload my code to [https://github.com/omgkenny6135/Fantasy-Pixel-Art-Net-Code/blob/main/Artnet_receiver_T5.ino](url), It is changed from `receive_fastled`, and it works very will.

hideakitai, 2021-01-12:
@omgkenny6135 Thanks, but I want to reproduce the compile error in my environment, so please give me your code based on `receive_fastled` which uses `FastLED`.

hideakitai, 2021-01-16:
@omgkenny613 Hi, the problem has solved? I want to reproduce the compile error if you can give me your code with errors.

hideakitai, 2021-01-20:
I'm going to close this issue but if you still have problem, please feel free to reopen this.

closed 2021-01-20 by hideakitai