hideakitai/ArtNet #15: Compile Error with Industruino

gunstr, 2021-01-17:
Hi,
I tried to compile the sender example for an Industruino, but I get a bunch of errors, It's a SAMD board and should be pretty similar to Arduino MKR Zero.  If I compile for the MKR Zero I get no errors.
First error is: "Compiler does not support __has_include, please report a bug against the ArxTypeTraits library about this."
I guess there are some difference in the board definitions that should be possible to fix, but it's beyond my level of knowledge to figure out what.
Any suggestion what I can do?

hideakitai, 2021-01-19:
@gunstr  Hi, Industruino is not supported currently. The workaround is to change

`ArtNet/util/ArxTypeTraits/ArxTypeTraits/has_include.h` Line 18

from

```C++
#elif defined(ARDUINO_SAM_DUE)
```

to

```C++
#elif defined(ARDUINO_SAM_DUE) || defined(ARDUINO_SAMD_INDUSTRUINO_D21G)
```

Maybe this works. Could you try it?

gunstr, 2021-01-19:
@hideakitai Thanks a lot, that worked like a charm!
Appreciate your fast response.

hideakitai, 2021-01-20:
That's good. Please let me know if you have any other problems.

closed 2021-01-20 by hideakitai