hideakitai/ArtNet #16: ART-NET 32 universos.

wilderpadilla, 2021-02-03:
hice unas modificaciones (muy simple) a tu librería art-net, para que pueda recibir 32 universos.
 ya que no estoy familiarizado con C++ y se me complica entender el funcionamiento de tu librería. 
estoy trabajando con ESP32 + w5500 transeiver ethernet + compilador Arduino.
también logre que ArtPoll devuelva los 32 universos.  
funciona bien y si recibe 32 universos. 
me gustaría entender mejor tu librería, si puedes brindarme una guía de el c++ que usas.
me gustaría que tu librería quedara con la opción de  recibir hasta los 32 universos.

hideakitai, 2021-02-24:
Previously this library accepts 32bit universe, but I've changed not to allow it (based on the Art-Net spec). But I also think this library should accept 32bit universe for flexibility. Currently I don't have time but I will do it if I have time.

hideakitai, 2021-03-26:
Supported again in v0.2.2

closed 2021-03-26 by hideakitai