hideakitai/ArtNet #17: esp32 + USR-ES1 W5500 Chip stoped working

LaseGraupenschwitz, 2021-02-11:
Hey,

I want to build an artnet Node with 2 Universes. So I have found your libary and it looked nice :)
The first I want to do is to get Artnet on the esp32 over ehternet. So I have an Wemos d1 mini esp32 and an USR-ES1 w5500 chip.

I have used your code example to show me the artnet data from uiverse0 on serial Monitor.

It worked well, but after a while (5-30 minutes) it stoped working. There are no error messages. I cant reach  the esp in network with ping.
I can´t imagine what the problem is. Maybe some failure in communication between the esp32 and the w5500?

[image]

I have no Idea how to recreate this error. 

Do you have some Ideas?

LaseGraupenschwitz, 2021-02-11:
I have also looked with Wireshark.
until no. 4676 - everything worked fine.

(2.0.0.71 = ESP, 2.0.0.10 = PC)
an after this the esp/w5500 stoped working :(

 so maybe is there a problem with to many network requests?
because before the 4676 only DMX channels where transmitted an no other protocoll are displayed.
[image]

Best Regards,
Mario Müller.

wilderpadilla, 2021-02-17:
hola, puedes poner el circuito esquemático,
y puedes poner el programa. que librería estas usando para el w5500.

wilderpadilla, 2021-02-17:
para que la librería de Hideakitai me funcione con el w5500 hice unos cambios pequeños en esta, ya qye no estaba la declaracion para usar el esp32 por confección SPI (solo para que se conecte por wifi).
y con la IDE de Arduino uso las siguientes librerías.
#include   <SPI.h>
#include <Ethernet.h>
#include <EthernetUdp.h>   

haciendo que funcione  el W5500 +ESP32 tranquilamente puedes recibir los 4 universos

hideakitai, 2021-02-24:
@LaseGraupenschwitz Sorry for the late reply, that seems to be a hang-up. Could you try the another firmware version of ESP? For example, if you use 3.x.x, please try 2.x.x and 1.x.x.

hideakitai, 2021-03-26:
I'm going to close this issue because I don't have experienced it. Please let me know if you still have the problem

closed 2021-03-26 by hideakitai