hideakitai/ArtNet #18: ESP32 boards with ethernet still use wifi

solexious, 2021-02-19:
https://github.com/hideakitai/ArtNet/blob/2f06b64242eaec4786433ef072332c3e869deb86/Artnet.h#L13-L19

When using esp32 boards with ethernet, i.e. https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE the library defaults to using wifi without a choice to use ethernet instead.

Is there a way to force using either the wifi or ethernet methods of this library irregardless of the libraries choice? If not currently, can the logic above be modified to allow overriding with a #define within the main sketch before the library is called?

wilderpadilla, 2021-02-19:
hola, la librería por defecto si encuentra un modulo con conexión wifi habilita el uso del wifi para ser usado.
ahora intente definir el uso de la librería de Ethernet en mi programa y no me funciono.
lo que hice es cambiar unas líneas de código, para que la librería solo trabaje por conexión Ethernet. la desventaja es que si quiero usar Wifi, tengo que revertir el cambio.

en mi caso uso el w5500  por (SPI)con la Ethernet  por conexión por SPI, y ha funcionado muy bien. en el modulo LiLYGO-T-ETH-POE que tienes hay que verificar que transeiver Ethernet usa, y hacer la pruebas.

 14...#define ARTNET_ENABLE_WIFI
cambie por
14....#define ARTNET_ENABLE_ETHER  //esto anula el uso de wifi para esta librería

teniendo en cuenta que para el w5500 huso la librería =   https://github.com/arduino-libraries/Ethernet

solexious, 2021-02-19:
Yes thanks. I'm just hoping this can be supported in the library without modification.

I'm currently looking at the differences between the Ethernet.h library used currently and the ETH.h library used by the LiLYGO-T-ETH-POE board to see if they are compatable.

wilderpadilla, 2021-02-19:
pero, tu quieres hacer un art-net ?
porque la librería de hideakitai es para hacer una conexión con protocolo ART-NET
si quieres hacer funcionar tu tarjeta con el transceiver LAN-7820, creo que no funcionara ya que la librería Ethernet.h, ya que  trabaja para conexión SPI, el LAN-7820 no se comunica por SPI.
si NO quieres hacer un art-net, te recomiendo abrir un nuevo hilo.

si (SI) quieres hacer  un ART-NET, confírmame si el código del transceiver es LAN-8720, y poder darte una mano en hacerlo funcionar.
/

solexious, 2021-02-20:
I'm sorry I think google translate is having an issue, I'm not sure I understand what you mean.

hideakitai, 2021-02-24:
@solexious Sorry for the late reply. Yes, currently only WiFi is supported for ESP. If you still want to use Ethernet on ESP, I can support Ethernet for ESP in a few days. But supporting ETH will take a long time, so I don't have an  idea to support ETH at the moment... I hope someone create the pull request to support ETH...

hideakitai, 2021-04-02:
@solexious v0.2.4 supports both WiFi and Ethernet on ESP32 and ESP8266

closed 2021-04-02 by hideakitai