hideakitai/ArtNet #20: Blocking latency with steppers

fredcazaux, 2021-03-30:
Hi @hideakitai !

First, thank you for this great library I use a lot in my projects !

I have an issue in a current project where I use steppers and need them to move while still receiving ArtNet to update LEDs.
I'm using accelstepper library to drive the steppers, and I use TMC2208 drivers. I use an Arduino Mega 2560. The artnet is sent from MagicQ on PC to the arduino by an ethernet shield.

As a test, I've written a simple sketch where a stepper is moving constantly, So in my loop(), I only have artnet.parse(); and stepper.run(); to get it as fast as possible. For this test, my callback function is empty.

When I run the sketch, my motor is running smoothly when I don't send any artnet packet to the arduino. But when I start sending packets, I can hear/feel/see the ~40FPS of the artnet blocking the stepper. As I have my callback function empty, I can tell this issue is coming from blocking functions in artnet or ethernet libraries.

Would you have any idea how to help this being smoother?

Thank you so much for your help, and congrats again for this huge library !

hideakitai, 2021-04-02:
Hi @fredcazaux , it may be the blocking by Ethernet library I think. You can test it without using ArtNet library by just receiving udp packet from MagicQ like this.

```C++
EthernetUdp udp;

void setup() {
    udp.begin(6454);

   // stepper stuff...
}

void loop() {
    const size_t size = udp.parsePacket();
    if (size != 0)  { // if udp packet has come
        uint8_t d[size];
        udp.read(d, size); // read and discard
    }

   // stepper stuff...
}
```

If it is the blocking is caused by Ethernet library, it may be better to `stepper.run()` in 40fps (not too fast). Please try it.

hideakitai, 2021-04-13:
I'm going to close this issue but if you have still problem, please feel free to comment.

closed 2021-04-13 by hideakitai