hideakitai/ArtNet #21: ESP32 Without ethernet

topjor, 2021-05-08:
I am currently using a lolin d32 ESP32 board with just wifi, but I keep getting the error that it can't find `Ethernet.h`
I am already using the deep mode for the library resolving (ldf mode)
I also have an ESP8266 but that is not giving any errors, so I may be missing a library (or the library finder) or there is a platform/board support issue.
There is also no way that I see to disable ethernet or wifi currently, if enabled, there is no way to disable.

hideakitai, 2021-05-24:
@topjor Supported in v0.2.8. Please confirm if it works!

closed 2021-05-24 by hideakitai

topjor, 2021-05-25:
I can confirm this works fine. Thank you.

(tried to check yesterday but PIO's caching system would not allow me to update sooner)