hideakitai/ArtNet #22: Use with ENC28J60 ethernet shield?

niallfleming, 2021-05-25:
Is it possible to swap out the Ethernet library used?

UIPEthernet works with the ENC28J60 chip, but the standard Ethernet lib does not.

Not sure if I'm being stupid or there's a way to override it?

hideakitai, 2021-05-26:
@niallfleming Hi, though it's not supported, maybe you can use it if UIPEthernet is compatible with the standard Ethernet library. Please try to replace `Artnet artnet;` with the following.

```C++
#include <Artnet.h>

// define your artnet manager with Eth
using ArtnetUIP = arx::artnet::Manager<UIPUDP>;
ArtnetUIP artnet;

// ...
```

niallfleming, 2021-05-26:
@hideakitai Thanks for that. Seems like it is still trying to load the normal library as well as I get duplicated class errors. Not sure if there's a way to avoid that.

I can get round it by commenting the Includes for Ethernet.h and EthernetUdp.h in ArtnetEther.h, and defining ESP8266 in order to avoid it attempting to call Ethernet.MACAddress(mac); in ArtnetCommon.

Which gets me to the point where it compiles, but then complains that it's using 135% of global variables, which I don't think is coming from my sketch, it has to be mainly from the includes I guess, probably the UIPEthernet lib.

hideakitai, 2021-05-27:
@niallfleming You can reduce the size of the library if you only need either receiving or sending the artnet.

```C++
#include <Artnet.h>

using ArtnetSenderUIP = arx::artnet::Sender<UIPUDP>;
using ArtnetReceiverUIP = arx::artnet::Receiver<UIPUDP>;

ArtnetReceiverUIP artnet;
// OR
ArtnetSenderUIP artnet;
```

niallfleming, 2021-05-27:
I've managed to get a lot further with this. It turns out that someone had done the work to make the UIPEthernet library more compatible with Ethernet - EthernetENC - and it seems a little smaller. So now it's compiling and uploading and pingable - it seems like the example MAC address listed in examples/Ethernet/receive_fastled. But I'm not seeing it as a node in MagicQ.

MagicQ is using broadcast by default for discovery, is that what this library expects and listens on? Is there somewhere I can add a judicious serial.println to check the broadcast it has computed.

I see the subscribe_net and subscribe_subnet can be passed to begin(), but I am not sure what these relate to by the name given and the expected values Net must be < 128, Subnet < 16.

I'm only testing (eventually) with 5 LEDS, so I've used an IP in the local subnet which is a /24 (255.255.255.0), so I expect packets on the broadcast address 192.168.0.255, which I do see coming from MagicQ:

21:03:58.645985 IP (tos 0x0, ttl 64, id 7015, offset 0, flags [none], proto UDP (17), length 558)
    192.168.0.101.6454 > 192.168.0.255.6454: [udp sum ok] UDP, length 530
21:03:59.066090 IP (tos 0x0, ttl 64, id 52887, offset 0, flags [none], proto UDP (17), length 42)
    192.168.0.101.6454 > 192.168.0.255.6454: [udp sum ok] UDP, length 14

niallfleming, 2021-05-27:
OK so I read the documentation on art-net.org.uk and I understand that net/subnet refer to artnet universe addressing. So the default of universe 0, net 0, subnet 0 should relate to Art-0 as configured in MagicQ.

hideakitai, 2021-05-28:
@niallfleming You can set `net` and `subnet` by using `artnet.begin(net, subnet)`. After that, please subscribe 4-bit universe with `artnet.subscribe(universe, callback)`.

Please check [here](https://github.com/hideakitai/ArtNet/blob/8adc0f088c73e70cf0aa21fc5c2f2d4ee8057b02/Artnet/ArtnetCommon.h#L260) if the packet has come. And if the ArtPoll packet has come, [this block](https://github.com/hideakitai/ArtNet/blob/8adc0f088c73e70cf0aa21fc5c2f2d4ee8057b02/Artnet/ArtnetCommon.h#L277) will be executed.

niallfleming, 2021-05-31:
Hi

I've got so much further! I've got it receiving on local broadcast at last, with some help from the current maintainer of EthernetENC, it turns out that it was filtering out everything but ARP... for some reason.

So now the board is receiving the ArtPoll 0x2000, and it definitely enters the poll_reply method but somewhere between the top and bottom of that method it's getting stuck, I've got a Serial.println just prior to `static const IPAddress local_broadcast_addr` 

There's 214 bytes left for global vars, could that be an issue?

hideakitai, 2021-06-01:
@niallfleming The struct `ArtPollReply` requires 239 bytes so maybe it is because of the memory shortage. Are you using the `Artnet` class? If you still using `Artnet` (integrated sender and receiver) and no need to send Artnet packets, please try to use `ArtnetReceiver`.

> @niallfleming You can reduce the size of the library if you only need either receiving or sending the artnet.
>
> ```c++
> #include <Artnet.h>
>
> using ArtnetSenderUIP = arx::artnet::Sender<UIPUDP>;
> using [...]

niallfleming, 2021-06-01:
No I'm only using the Receiver as shown above, the code in my sketch is largely unchanged except that I removed a bunch of serial.print(ln)s, and reenabled all the FastLED stuff.

I'm starting to think this may be impossible on nano.

hideakitai, 2021-06-01:
Yes, I recommend you to use the more powerful board if you can (Teensy, ESP32, etc.).

hideakitai, 2021-06-04:
@niallfleming I'm going to close this issue but please reopen if you have any other suggestions.

closed 2021-06-04 by hideakitai