hideakitai/ArtNet #23: Documentation concerning sending Artnet packets

C0pernic, 2021-07-20:
Hi there !

I use the Artnet library for teensy 4.1 but I got stuck with sending packets to a Artnet device. Is there any function to format Artnet messages ? 

Basically, I need to send DMX datas through Artnet but I don't get how to format header and stuff in the library.

Thanks in advance,

hideakitai, 2021-07-20:
Hi, you can find how to stream packets in examples https://github.com/hideakitai/ArtNet/blob/master/examples/Ethernet/sender/sender.ino

Or you can send a packet once as follows as shown in README.

```
artnet.send("127.0.0.1", universe15bit, data_ptr, size); // one-line send
```

Header bytes are automatically added to DMX data.

hideakitai, 2021-07-26:
@C0pernic If you still have the problem, please reopen the issue.

closed 2021-07-26 by hideakitai