hideakitai/ArtNet #24: Compiling receiver example on Raspberry pi pico rp2040

cgiles, 2021-08-07:
Hello,
I'm trying to compile your example receiver on my rp2040, and I got this error message : 

> In file included from D:\Users\User\Documents\Arduino\libraries\ArtNet/ArtnetEther.h:14:0,
                 from D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:3:
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h: In instantiation of 'std::enable_if_t<arx::is_callable<F>::value> arx::artnet::Receiver_<S>::subscribe(uint8_t, F&&) [with F = setup()::<lambda(const uint8_t*, uint16_t)>; S = EthernetUDP; std::enable_if_t<arx::is_callable<F>::value> = void; uint8_t = unsigned char]':
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:39:6:   required from here
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:345:32: error: no matching function for call to 'setup()::<lambda(const uint8_t*, uint16_t)>::__lambda5(const char [19])'
                 Serial.println(F("too many callbacks"));
                                ^~~~~~~~~~~~~~~~~~~~~~~
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note: candidate: constexpr setup()::<lambda(const uint8_t*, uint16_t)>::<lambda>(const setup()::<lambda(const uint8_t*, uint16_t)>&)
     artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
                                   ^
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note:   no known conversion for argument 1 from 'const char [19]' to 'const setup()::<lambda(const uint8_t*, uint16_t)>&'
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note: candidate: constexpr setup()::<lambda(const uint8_t*, uint16_t)>::<lambda>(setup()::<lambda(const uint8_t*, uint16_t)>&&)
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note:   no known conversion for argument 1 from 'const char [19]' to 'setup()::<lambda(const uint8_t*, uint16_t)>&&'
In file included from D:\Users\User\Documents\Arduino\libraries\ArtNet/ArtnetEther.h:14:0,
                 from D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:3:
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:348:36: error: no matching function for call to 'setup()::<lambda(const uint8_t*, uint16_t)>::__lambda5(const char [23])'
                     Serial.println(F("universe out of bounds"));
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note: candidate: constexpr setup()::<lambda(const uint8_t*, uint16_t)>::<lambda>(const setup()::<lambda(const uint8_t*, uint16_t)>&)
     artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
                                   ^
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note:   no known conversion for argument 1 from 'const char [23]' to 'const setup()::<lambda(const uint8_t*, uint16_t)>&'
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note: candidate: constexpr setup()::<lambda(const uint8_t*, uint16_t)>::<lambda>(setup()::<lambda(const uint8_t*, uint16_t)>&&)
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:28:35: note:   no known conversion for argument 1 from 'const char [23]' to 'setup()::<lambda(const uint8_t*, uint16_t)>&&'
In file included from D:\Users\User\Documents\Arduino\libraries\ArtNet/ArtnetEther.h:14:0,
                 from D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:3:
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h: In instantiation of 'std::enable_if_t<arx::is_callable<F>::value> arx::artnet::Receiver_<S>::subscribe(uint8_t, F*) [with F = void(const unsigned char*, short unsigned int); S = EthernetUDP; std::enable_if_t<arx::is_callable<F>::value> = void; uint8_t = unsigned char]':
D:\Users\User\AppData\Local\Temp\arduino_modified_sketch_849912\receiver.ino:42:41:   required from here
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:360:32: error: invalid cast to function type 'void(const unsigned char*, short unsigned int)'
                 Serial.println(F("too many callbacks"));
                                ^~~~~~~~~~~~~~~~~~~~~~~
D:\Users\User\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:363:36: error: invalid cast to function type 'void(const unsigned char*, short unsigned int)'
                     Serial.println(F("universe out of bounds"));
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Erreur de compilation pour la carte Raspberry Pi Pico

have you got a fix for this ? 
Thanks for your work,
regards

hideakitai, 2021-08-09:
Hi, thank you for reporting. This is because Pi Pico doesn't support the `F()` macro. You can compile it if you remove the `F()` macro from `ArtnetCommon.h`. Could you try it, please?

hideakitai, 2021-08-10:
Ok, I've fixed it in v0.2.9. `F()` is not macro but just a templated function in the pico core library.

https://github.com/arduino/ArduinoCore-mbed/blob/0e57d7b77e343978b7f202796a28568538ac2776/cores/arduino/Arduino.h#L31-L38

So the template name `F` has conflicted with the function `F()`... Could you try v0.2.9, please?

hideakitai, 2021-08-15:
@cgiles If you still have the problem, please let me know.

closed 2021-08-15 by hideakitai