hideakitai/ArtNet #26: Problem when receiving multiple universes

damienrivoal, 2021-11-13:
Hi,

Thanks a lot for your work !

I have a problem when receiving more than 3 universes at the same time.

Every 1 universe out of 4 is missed.
e.g. if I receive at the same time universes from 0 to 15, it's always missing universes 3, 7, 11 and 15. If I receive at the same time universes from 3 to 18, it's always missing universes 6, 10, 14 and 18.

Any clue ?

When I monitor the signal out of my GrandMa 2 with Wireshark, I see every packet for every universe. But the arduino mega can't see it.

If I send universes one at a time, no problem at all, the arduino sees everything...

Really odd !!

Thank for your answer !

Best regards,

Damien Rivoal

damienrivoal, 2021-11-18:
Ok, after some research I found that the problem has to do with the hardware AND software. The Ethernet shield V1 I used works on a chipset W5100 which has only 16k buffer memory (8 for RX, 8 for TX). By default, the Ethernet.h library sets a buffer value for each socket to 2048 bytes. Every universe takes 530 bytes. Thus every fourth universe drops because the buffer gets full.

There's a way to increase buffer memory within the limit of hardware capacity. I found a solution here : https://github.com/lathoub/Arduino-AppleMIDI-Library/wiki/Enlarge-Ethernet-buffer-size-to-avoid-dropping-UDP-packages

I reduced the number of sockets to 2 by modifying Ethernet.h according to the link above. This way I got 15 universes running well.

The other option is to use an Ethernet shield V2 which works on a W5500 chipset. This way you get 32K memory instead of 16K.

Combining these two solutions, you should be able to get something like 30 universes running well.

Hope this can I help someone like me who's working with a quite large number of universes with a small piece of hardware ;)

Cheers

hideakitai, 2021-12-17:
@damienrivoal 
Sorry for the delay in replying. I've been busy with work and haven't been able to check, but I will do so when I have time.

hideakitai, 2022-01-02:
@damienrivoal Thank you for sharing the excellent knowledge!

closed 2022-01-02 by hideakitai

damienrivoal, 2022-01-02:
@hideakitai Thanks for taking time to answer 😉