hideakitai/ArtNet #27: Unsupported OpCode: 2100

mediaMicha, 2021-12-09:
Hello,
at first, thank you, that you build up this great library. 
I use your ArtNet-0.2-2.11 lib at my project as ArtNet in and out. In this project I control a media server to do a single layer transition. It’s work fine, if I use a normal ArtNet node sender to send ArtNet Data to my Arduino Mega with W5100 Ethernet shield. But if I connect it the network cable with GrandMA3 with GrandMA2 Software, so I become over the Arduino serial monitor, this error massage:
Unsupported OpCode: 2100
And my project don’t work. If I use your receiver example, the same error.

I hope someone can help me here. Thanks in advance.

I use for coding Arduino IDE 1.8.16
ArtNet send with: AVR Node - DMX Ethernet Gateway [https://www.ulrichradig.de/home/index.php/dmx/alias-2](url)
arduino mega 256 board
W5100 Mini LAN Ethernet Shield [https://www.ebay.de/itm/273892669982?chn=ps&_trkparms=ispr%3D1&amdata=enc%3A1GrljCyDWTMyVKz1BQ9SeYQ40&norover=1&mkevt=1&mkrid=707-134425-41852-0&mkcid=2&itemid=273892669982&targetid=1396098981494&device=c&mktype=pla&googleloc=9041941&poi=&campaignid=14472331666&mkgroupid=125370287694&rlsatarget=pla-1396098981494&abcId=9300652&merchantid=113833088&gclid=Cj0KCQiAzMGNBhCyARIsANpUkzNOON-QHsRqEhtsnski9ZbJnvpWZB-pL_t4zC9ATVQ5muC7F3XGEUYaAiBmEALw_wcB](url)
grandMA3 light with grandMA2 Software [https://www.malighting.com/product/grandma3-light-4010502/](url)

JohnLilley, 2021-12-12:
found similar problem when the network has 2 or more ArtNet nodes built with this library in it.
OpCode 2100 is POLL_REPLY.
In the Parse function of the library there is a switch statement that selects opcodes DMX & POLL and any others including POLL_REPLY are considered as unsupported opcodes.

Jack079, 2021-12-15:
Hello, 

I just started to try this library.
I used the example code Ethernet receiver and tried multiple software to try sending artnet.
I same massages.

When the software starts I get this message:
Unsupported OpCode: 8000
When I don't receive updates over artnet I get this message:
Unsupported OpCode: 2100
When I get data over artnet I get this message:
Unsupported OpCode: 5200

Hardware: 
- Arduino Nano 
- W5500 ethernet shield

Software:
- Onyx
-  DMXControl 3

lcava000, 2021-12-16:
Hi, have you the connection or pinout? thanks you

Jack079, 2021-12-16:
w5500 - Nano
SCNn   - 10
MOSI   - 11
MISO   - 12
SCLK    - 13

hideakitai, 2021-12-17:
@mediaMicha @JohnLilley @Jack079 
Hi, "unsupported opcode XXXX" is not an error message, it does not affect the sending and receiving of DMX data on ArtNet. v0.2.12 has made it so that such messages are not displayed by default, so please try it. `receiver.verbose(true)` to display it again.

JohnLilley, 2021-12-17:
Thanks for the speedy update, was no real problem to me other than regular lines appearing on the serial monitor.
Good library performance wise - i'm creating ArtNet nodes to run LED outdoor lighting and effects with relatively small processors (Arduino Mega) and soon ESP32's.  Thank you for your efforts.

mediaMicha, 2021-12-17:
HI, 
thank you for your fast fix. It's work fine with W5100 and W5500.

mediaMicha, 2021-12-19:
I'm very happy with this fix. Thank you

closed 2021-12-19 by mediaMicha