hideakitai/ArtNet #28: Image of connections

lcava000, 2021-12-16:
Thank you for your contribution.
But can you post an image of the connections with ESP32 and RS485 to the DMX? Thank you

hideakitai, 2021-12-17:
You can't use ArtNet with RS485. Maybe related: https://arcattack.com/dmx-over-rs485-with-an-arduino

closed 2021-12-17 by hideakitai

lcava000, 2021-12-17:
I would like to use ESP32 as an ART-NET receiver.
I connected to the esp32 according to these links.

  - RS-485 -
DI -> TX2 UART
DE -> GND (Receive Mode)
RE -> GND (Receive Mode)
RE -> RX2 UART
VCC -> VCC
GND -> GND
A -> DATA + DMX
B -> DATA - DMX

Why i can't use ArtNet with RS485?
Below I found a wiring diagram for arduino. 

[image]

hideakitai, 2021-12-17:
See the difference between DMX and ArtNet https://imimot.com/blog/lighting-for-beginners-dmx-and-art-net/