hideakitai/ArtNet #30: Change Artnet node name programmatically?

tomtomhoward, 2022-01-29:
Hi
Apologies for a possible feature request as I know you must be inundated and your library is fantastic.
Is there way to change the node name without editing the library?
I have multiple nodes running and it would be great to be able to identify between them. If I could change the name in the sketch it would be a bit help as I can change via editing the library but I have to do it every time between programming different nodes.
Apologies if there is already a way to do this and I haven't found it 
Many thanks
Tom

hideakitai, 2022-01-29:
Hi, have you tried ArtPollReplySetting?

> You can configure the information of ArtPollReply as follows
> void shortname(const String& sn)
> void longname(const String& ln)
> void nodereport(const String& nr)

tomtomhoward, 2022-01-29:
Apologies but how would you code that from within Arduino IDE? 
artnet.ArtPollReplySetting(name) or similar?

hideakitai, 2022-01-29:
Please call the methods listed above. First, please try it.

hideakitai, 2022-01-29:
artnet.shortname("foo");

tomtomhoward, 2022-01-29:
Thanks so much for that I couldn't get it right I was using artnet.short_name("")
Sorry for wasting your time I've found it in the readme, I was searching short_name as that's the default string in the library so I couldn't find it by searching. Many thanks, I have the hostname of the devices set and just want to duplicate it onto the artnet poll.

I am reusing some old analogue pre-DMX dimmers by fitting ESPs to fire the triacs and control them over artnet.
I am a theatre technician by trade building stages and carpentry so apologies a lot of this is not natural to me

hideakitai, 2022-01-29:
No problem. I hope your project goes well!

closed 2022-03-07 by hideakitai