hideakitai/ArtNet #33: Stuck when more then 2 Universes are sent

nbastijans, 2022-03-10:
Hey,
first of all great project.

When i'm sending signals through example madrix, the pixels work fine when i'm just sending 2 universes to the network.
Just in the moment i'm sending 3 universes through the network, the pixels are getting slower, and the images getting stuck.

In my program code i only subscribed 2 universes, but it seems to process the third, fourth ... universe.
Would it be possibe to filter it before processing it?

hideakitai, 2022-03-11:
@nbastijans Hi, 

Could you give me more information to reproduce this issue?

- IDE (Arduino IDE, PlatformIO, etc.) : Arduino IDE ?
- Board/MCU: ESP32-WROOM32 ?
- Board Firmware Version: v2.0.2 ?
- Library Version: ArtNet v0.2.12 ?
- Reproducable & Minimal & Compilable Code
- Additional Information (System chart, etc.)

fastfourier666, 2022-07-25:
Hi, I'm also having this issue:

- Arduino 1.8.19, Library version latest as of June 2022
- ESP32 WROOM32, WZ5500

I am sending unicast Artnet from Jinx! which is fixed at 25fps. 

It seems like the problem is the size of the universes. If I send univ. 16, 17, 18 and 19, number 19 only receives the occasional frame, 1 or 2 every 10 seconds.

If I reduce the number of channels in universe 18 to a lower number (1 or 10) then I get 25 frames/sec from all.

```
#include <ArtnetEther.h>

#define RESET_P 21        // rset line for ethernet module

uint8_t eth_MAC[] = { 0x02, 0xF7, 0x0D, 0xBE, 0xEF, 0x01 };
ArtnetReceiver artnet;

uint8_t frc[24];
uint32_t secTimer=0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(200);
  Serial.printf ("\nhello\n");
  Ethernet.init(5);
  WizReset();
  Ethernet.begin(eth_MAC);
  Serial.print("Ethernet IP is: ");
  Serial.println(Ethernet.localIP());

  artnet.begin(0, 1);
  artnet.subscribe ([&](uint16_t u, const uint8_t* data, const uint16_t size) {
    frc[u]++;     // increase frame counter for this universe
  });
}

void loop() {
  // put your main code here, to run repeatedly:
  artnet.parse();
  if (millis() - secTimer > 1000) {
    secTimer = millis();
    Serial.printf ("fps_univ: %i %i %i %i %i %i\n", frc[16], frc[17], frc[18], frc[19], frc[20]);
    for (int i = 0; i < 24; i++) frc[i] = 0;
  }
}

void WizReset() {
  Serial.print("Resetting Wiz W5500 Ethernet Board...  ");
  pinMode(RESET_P, OUTPUT);
  digitalWrite(RESET_P, HIGH);
  delay(250);
  digitalWrite(RESET_P, LOW);
  delay(50);
  digitalWrite(RESET_P, HIGH);
  delay(350);
  Serial.println("Done.");
}
```

fastfourier666, 2022-07-25:
It's an issue with the Ethernet library. I wrote a quick and dirty Artnet receiver using just EthernetUDP, and got the same effects.

In Ethernet.h you have to uncomment this line (48). There is even a comment about Artnet in the blurb above it!

`//#define ETHERNET_LARGE_BUFFERS`

Just above this there is a section which defines `MAX_SOCK_NUM`. The WZ5500 can do 8 sockets but to make the large buffers work you have to define this number as <8. I just commented out everything except the line:

`#define MAX_SOCK_NUM 4`

Since I only need to receive Artnet I only needed one socket anyway. I think it would be possible to make it work with 8 sockets but it would need a bit more "finessing" in the rest of the library.

hansvana, 2022-08-11:
@fastfourier666 Thank you for your reply. This helped me improve reliability. Instead of freezing after a couple of seconds my program now runs for much longer with 2 universes receiving about 50 packages per second from MagicQ PC. 

However after running about 15 minutes it still suddenly froze (as in, no new data is printed to the serial port). Resetting makes everything work as normal again. I wonder if this is the same issue and if so, if there is a way to detect problems and maybe dump the current package and move on to the next.

- Arduino 1.8.13
- ESP32-WROOM32 WZ5500
- ArtNet v0.2.12
- Running [examples](https://github.com/hideakitai/ArtNet/tree/master/examples)/[Ethernet](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet)/[receiver](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet/receiver)/receiver.ino (modified to Serial print on both universes)

fastfourier666, 2022-08-11:
The example you are running prints all the channel values as text to the serial port, which is a lot. Your main loop is probably not executing fast enough, and the buffer is filling up before it gets back to `artnet.parse()` again.

Try changing the for loop (line 34 of the example) to only print the first couple of channels of each universe.

hansvana, 2022-08-11:
@fastfourier666 thanks for your reply! I had failed to mention in my previous post that I already tried exactly what you suggest. Truncated code below. I will move forward with my project and try and see if sending the received data verbatim to dmx also crashes the esp32, or if that is easier for it to do. I would still like to be able to detect buffer problems so maybe the esp can take action before it crashes.

```
    artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("artnet data (universe : ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < min(6, int(size)); ++i) { // <== only output the first 6 values
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });

    artnet.subscribe(universe2, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("artnet data (universe : ");
        Serial.print(universe2);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < min(6, int(size)); ++i) { // <== only output the first 6 values
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });
```

fastfourier666, 2022-08-11:
It looks like that should work, but I would go a step further and get rid of all the serial print stuff. Just have an LED respond to universe 1, channel 1 and blink it from your DMX software so you can tell it's still working. If you do serial.print anything, just keep a frame count and spit it out every second. Maybe also keep a count of each time `artnet.parse()` gets called every second. See if you can get it to fall over sooner by firing more data at it - either up the fps, or send more universes.

The W5500 docs suggest that incoming UDP datagrams will just get ignored if the buffer is full, so the freezing behaviour is weird. The chip might be getting into a state that the library doesn't expect - is all the power OK?

In general, using the ESP32 hardware watchdog is a good shout, although of course it doesn't address your problem.

hideakitai, 2023-08-01:
Thanks for your excellent knowledge. I am closing this issue.

closed 2023-08-01 by hideakitai