hideakitai/ArtNet #34: Arduino uno r3, w5500 ws8212 pin3 don't working

SilviuART, 2022-03-14:
[ArtNet](https://github.com/hideakitai/ArtNet)/[examples](https://github.com/hideakitai/ArtNet/tree/master/examples)/[Ethernet](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet)/[receive_fastled](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet/receive_fastled)/receive_fastled.ino

please help me!

Lightjams detect node but all led is off.
Sorry for my english

hideakitai, 2022-03-14:
Hi, WS2812? Please try FastLED without ArtNet first. I suspect it is hardware noise.

SilviuART, 2022-03-15:
with fast led it is ok, working, with [ArtNet](https://github.com/hideakitai/ArtNet)/[examples](https://github.com/hideakitai/ArtNet/tree/master/examples)/[Ethernet](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet)/[receiver](https://github.com/hideakitai/ArtNet/tree/master/examples/Ethernet/receiver)/receiver.ino it is ok, on com write artnet data.

thank you for response

hideakitai, 2022-03-15:
Thanks, could you give me your complete code of FastLED without ArtNet?

SilviuART, 2022-03-16:
#include <FastLED.h>

#define LED_PIN     3
#define NUM_LEDS    64
#define BRIGHTNESS  50
#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

// This example shows several ways to set up and use 'palettes' of colors
// with FastLED.
//
// These compact palettes provide an easy way to re-colorize your
// animation on the fly, quickly, easily, and with low overhead.
//
// USING palettes is MUCH simpler in practice than in theory, so first just
// run this sketch, and watch the pretty lights as you then read through
// the code.  Although this sketch has eight (or more) different color schemes,
// the entire sketch compiles down to about 6.5K on AVR.
//
// FastLED provides a few pre-configured color palettes, and makes it
// extremely easy to make up your own color schemes with palettes.
//
// Some notes on the more abstract 'theory and practice' of
// FastLED compact palettes are at the bottom of this file.

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    
    currentPalette = RainbowColors_p;
    currentBlending = LINEARBLEND;
}

void loop()
{
    ChangePalettePeriodically();
    
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; /* motion speed */
    
    FillLEDsFromPaletteColors( startIndex);
    
    FastLED.show();
    FastLED.delay(1000 / UPDATES_PER_SECOND);
}

void FillLEDsFromPaletteColors( uint8_t colorIndex)
{
    uint8_t brightness = 255;
    
    for( int i = 0; i < NUM_LEDS; ++i) {
        leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness, currentBlending);
        colorIndex += 3;
    }
}

// There are several different palettes of colors demonstrated here.
//
// FastLED provides several 'preset' palettes: RainbowColors_p, RainbowStripeColors_p,
// OceanColors_p, CloudColors_p, LavaColors_p, ForestColors_p, and PartyColors_p.
[... 85 lines cut ...]
    CRGB::Black,
    CRGB::Black
};

// Additional notes on FastLED compact palettes:
//
// Normally, in computer graphics, the palette (or "color lookup table")
// has 256 entries, each containing a specific 24-bit RGB color.  You can then
// index into the color palette using a simple 8-bit (one byte) value.
// A 256-entry color palette takes up 768 bytes of RAM, which on Arduino
// is quite possibly "too many" bytes.
//
// FastLED does offer traditional 256-element palettes, for setups that
// can afford the 768-byte cost in RAM.
//
// However, FastLED also offers a compact alternative.  FastLED offers
// palettes that store 16 distinct entries, but can be accessed AS IF
// they actually have 256 entries; this is accomplished by interpolating
// between the 16 explicit entries to create fifteen intermediate palette
// entries between each pair.
//
// So for example, if you set the first two explicit entries of a compact 
// palette to Green (0,255,0) and Blue (0,0,255), and then retrieved 
// the first sixteen entries from the virtual palette (of 256), you'd get
// Green, followed by a smooth gradient from green-to-blue, and then Blue.

hideakitai, 2022-03-16:
Your LED_TYPE is different from my example. Could you try to change it? Also note that my example drives only one pixel.

SilviuART, 2022-03-16:
help to code for Artnet arduino uno to ws8212

SilviuART, 2022-03-16:
problem is in line code:
FastLED.addLeds<NEOPIXEL, PIN_LED_DATA>(leds, NUM_LEDS);

hideakitai, 2022-03-16:
Please change NEOPIXEL to WS2811 if your FastLED code works fine.

SilviuART, 2022-03-16:
I modified code to:
#include <FastLED.h>  // include FastLED *before* Artnet

// Please include ArtnetEther.h to use Artnet on the platform
// which can use both WiFi and Ethernet
// #include <ArtnetEther.h>
// this is also valid for other platforms which can use only Ethernet
#include <Artnet.h>

// Ethernet stuff
const IPAddress ip(192, 168, 1, 181);
uint8_t mac[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB};

ArtnetReceiver artnet;
uint32_t universe1 = 1;  // 0 - 15
uint32_t universe2 = 2;  // 0 - 15

void callback(const uint8_t* data, const uint16_t size) {
    // you can also use pre-defined callbacks
}

// FastLED
#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
const uint8_t PIN_LED_DATA = 3;

void setup() {
    Serial.begin(115200);

    

    Ethernet.begin(mac, ip);
    artnet.begin();
    // artnet.subscribe_net(0);     // optionally you can change
    // artnet.subscribe_subnet(0);  // optionally you can change

    // if Artnet packet comes to this universe, this function is called
    artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("artnet data (universe : ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < size; ++i) {
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });

    // you can also use pre-defined callbacks
    artnet.subscribe(universe2, callback);
    //FastLED.addLeds<NEOPIXEL, PIN_LED_DATA>(leds, NUM_LEDS);
}

void loop() {
    artnet.parse();  // check if artnet packet has come and execute callback
    FastLED.show();
}

and serial monitor it is ok working_ but not leds

SilviuART, 2022-03-16:
how to dend signal  serial port to arduino pins

hideakitai, 2022-03-16:
You should delete comment 
//FastLED.addLeds<NEOPIXEL, PIN_LED_DATA>(leds, NUM_LEDS);
and change to 
FastLED.addLeds<WS2811, PIN_LED_DATA>(leds, NUM_LEDS);

SilviuART, 2022-03-16:
#include <FastLED.h>  // include FastLED *before* Artnet

// Please include ArtnetEther.h to use Artnet on the platform
// which can use both WiFi and Ethernet
// #include <ArtnetEther.h>
// this is also valid for other platforms which can use only Ethernet
#include <Artnet.h>

// Ethernet stuff
const IPAddress ip(192, 168, 1, 181);
uint8_t mac[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB};

ArtnetReceiver artnet;
uint32_t universe1 = 1;  // 0 - 15
uint32_t universe2 = 2;  // 0 - 15

void callback(const uint8_t* data, const uint16_t size) {
    // you can also use pre-defined callbacks
}

// FastLED
#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
const uint8_t PIN_LED_DATA = 3;

void setup() {
    Serial.begin(115200);
    //FastLED.addLeds<WS2811, PIN_LED_DATA>(leds, NUM_LEDS);
    

    Ethernet.begin(mac, ip);
    artnet.begin();
    // artnet.subscribe_net(0);     // optionally you can change
    // artnet.subscribe_subnet(0);  // optionally you can change
    // if Artnet packet comes to this universe, this function is called
    artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("artnet data (universe : ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < size; ++i) {
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });

    // you can also use pre-defined callbacks
    artnet.subscribe(universe2, callback);
    
}

void loop() {
    artnet.parse();  // check if artnet packet has come and execute callback
    FastLED.show();
}

working serial monitor, 

#include <FastLED.h>  // include FastLED *before* Artnet

// Please include ArtnetEther.h to use Artnet on the platform
// which can use both WiFi and Ethernet
// #include <ArtnetEther.h>
// this is also valid for other platforms which can use only Ethernet
#include <Artnet.h>

// Ethernet stuff
const IPAddress ip(192, 168, 1, 181);
uint8_t mac[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB};

ArtnetReceiver artnet;
uint32_t universe1 = 1;  // 0 - 15
uint32_t universe2 = 2;  // 0 - 15

void callback(const uint8_t* data, const uint16_t size) {
    // you can also use pre-defined callbacks
}

// FastLED
#define NUM_LEDS 64
CRGB leds[NUM_LEDS];
const uint8_t PIN_LED_DATA = 3;

void setup() {
    Serial.begin(115200);
    FastLED.addLeds<WS2811, PIN_LED_DATA>(leds, NUM_LEDS);
    

    Ethernet.begin(mac, ip);
    artnet.begin();
    // artnet.subscribe_net(0);     // optionally you can change
    // artnet.subscribe_subnet(0);  // optionally you can change
    // if Artnet packet comes to this universe, this function is called
    artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("artnet data (universe : ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < size; ++i) {
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });

    // you can also use pre-defined callbacks
    artnet.subscribe(universe2, callback);
    
}

void loop() {
    artnet.parse();  // check if artnet packet has come and execute callback
    FastLED.show();
}

serial monitor not working, what it is hapening

SilviuART, 2022-03-16:
thank you, working with 14 led.

hideakitai, 2022-03-16:
Because Uno does not have enough memory, the number of LEDs is limited.

closed 2022-03-16 by hideakitai

SilviuART, 2022-03-16:
Thank you for your help, will I try Teensy 4.1, another suggestion?

hideakitai, 2022-03-16:
ESP32 and Raspberry Pi Pico are also good choices, I think.