hideakitai/ArtNet #38: Artnet not sending when using WiFi.softAP(ssid, password) method

DrBit, 2022-08-24:
Hi there

I've encountered a problem with the library

I'm using ESP8266 as an access point so another ESP8266 can conect to it. 
When using ESP8266 as AP turns out it does not transmit any artnet data. Same exaple works perfect as wifi client.

here my code to test:

#include <elapsedMillis.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>

#ifndef APSSID
#define APSSID "ESPap"
#define APPSK  "thereisnospoon"
#endif

/* Set these to your desired credentials. */
const char *ssid = APSSID;
const char *password = APPSK;

// Please include ArtnetWiFi.h to use Artnet on the platform
// which can use both WiFi and Ethernet
#include <ArtnetWiFi.h>
// this is also valid for other platforms which can use only WiFi
// #include <Artnet.h>

// WiFi stuff
const IPAddress ip(192, 168, 2, 2);
const IPAddress gateway(192, 168, 2, 1);
const IPAddress subnet(255, 255, 255, 0);

ArtnetWiFiSender artnet;
const String target_ip = "192.168.4.2";
uint32_t universe = 1;

const uint16_t size = 512;
uint8_t data[size];
uint8_t value = 0;

void setup() {
    Serial.begin(115200);
    Serial.println("\nArtnet serial test started");

    setup_ap();

    pinMode(2, OUTPUT);
    pinMode(16, OUTPUT);

    artnet.begin();
    // artnet.begin(net, subnet);     // optionally you can change net and subnet
}

void setup_ap() {
  Serial.print("Setting soft-AP configuration ... ");
  Serial.println(WiFi.softAPConfig(ip, gateway, subnet) ? "Ready" : "Failed!");

  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP(ssid, password) ? "Ready" : "Failed!");

  Serial.print("Soft-AP IP address = ");
  Serial.println(WiFi.softAPIP());
}

void loop() {
    loop_ap_info();

    // Set all universe same value
    int value2 = (millis() / 4) % 256;
    memset(data, value2, size);

    // if (data[0] != value) {
        data[0] = value;        // set first chanel equals value
        artnet.streaming_data(data, size);
        artnet.streaming(target_ip, universe);  // automatically send set data in 40fps
        // artnet.streaming(target_ip, net, subnet, univ);  // or you can set net, subnet, and universe
    // }
}

int lastStationsConected = -1;
elapsedMillis checkClientsTimer;
elapsedMillis ledToggleTimer;

void loop_ap_info() {
    if (checkClientsTimer > 1500) {
        checkClientsTimer = 0;
        int newStationsConnected = WiFi.softAPgetStationNum();
        if (lastStationsConected != newStationsConnected) {
            lastStationsConected = newStationsConnected;
            Serial.printf("Stations connected to soft-AP = %d\n", newStationsConnected);
        }
    }

    if (lastStationsConected > 0) {
        // blink led to signal we got client connected
        if (ledToggleTimer > 300) {
            toggleLed (2);
            toggleLed (16);
            ledToggleTimer = 0;
        }
    }
}

void toggleLed (int pinLedToToggle) {
  digitalWrite (pinLedToToggle, !digitalRead(pinLedToToggle));
}

marcobrianza, 2023-04-28:
I have the same issue

thinksilicon, 2023-07-06:
Hey,
I'd like to bump this issue as well... it seems to work fine when running as regular wifi client, it doesn't seem to receive or send anything when running as HOST_AP/softAP. Using an ESP32 Wroom module.

hideakitai, 2023-08-14:
Fixed by https://github.com/hideakitai/ArtNet/pull/45

closed 2023-08-14 by hideakitai