hideakitai/ArtNet #44: Olimex ESP32-EVB, ethernet and Artnet does it work?

MikeZomo, 2023-06-16:
Hi,

I have ben testing out ArtNet on the  Olimex ESP32-EVB  (which comes with a RJ45) using  you the demo  code supplied but I am getting no response. I am  primarily interested Artnet  over ethernet. I use external test  equipment to generate Artnet packets and other equipment to to verify Artnet to verify  that the packets are being received. On the same network nothing is being received by the ESP32.
I verified with  the boards  works using Ethernet by setting up  a web client on it.  The board has a  LAN8720 on it. Any help you can provide would  be much appreciated.

Mike Zomo

MikeZomo, 2023-06-16:
Hi,

Does  your library support some means  of being identified on the ArtNet network?

Mike Zomo

MaxWimmreuter, 2023-07-23:
Hi,

as far as I know, boards that require the ETH.h library (like Olimex Boards) do not work.

But there is a fork from burner- which adds support for the ETH.h library:

[https://github.com/burner-/ArtNet](https://github.com/burner-/ArtNet)

Here an example with static IP for an Olimex Board:
I can receive ArtNet from MA onPC and the device shows up as a node in The ArtNetominator.

```
#include <Arduino.h>
#include "ArtnetETH.h"
#include "Artnet/ArtnetCommon.h"

static bool eth_connected = false;
ArtnetWiFiReceiver artnet;
uint32_t universe1 = 0; // 0 - 15

void WiFiEvent(WiFiEvent_t event)
{
    switch (event)
    {
    case ARDUINO_EVENT_ETH_START:
        Serial.println("ETH Started");
        // set eth hostname here
        ETH.setHostname("esp32-ethernet");
        break;
    case ARDUINO_EVENT_ETH_CONNECTED:
        Serial.println("ETH Connected");
        break;
    case ARDUINO_EVENT_ETH_GOT_IP:
        Serial.print("ETH MAC: ");
        Serial.print(ETH.macAddress());
        Serial.print(", IPv4: ");
        Serial.print(ETH.localIP());
        if (ETH.fullDuplex())
        {
            Serial.print(", FULL_DUPLEX");
        }
        Serial.print(", ");
        Serial.print(ETH.linkSpeed());
        Serial.println("Mbps");
        eth_connected = true;
        break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
        Serial.println("ETH Disconnected");
        eth_connected = false;
        break;
    case ARDUINO_EVENT_ETH_STOP:
        Serial.println("ETH Stopped");
        eth_connected = false;
        break;
    default:
        break;
    }
}

void setup()
{
    Serial.begin(115200);
    WiFi.onEvent(WiFiEvent);

    ETH.begin();
    ETH.config(IPAddress(10, 0, 0, 100),IPAddress(10, 0, 0, 1),IPAddress(255, 0, 0, 0),IPAddress(10, 0, 0, 1), IPAddress(10, 0, 0, 1));

    artnet.begin();

    delay(100);

    Serial.println("Started!");

    artnet.subscribe(universe1, [&](const uint8_t* data, const uint16_t size) {
        Serial.print("lambda : artnet data (universe : ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < size; ++i) {
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });
}

void loop()
{
    artnet.parse();
}
```

hideakitai, 2023-08-13:
Fixed by https://github.com/hideakitai/ArtNet/pull/48

closed 2023-08-13 by hideakitai