hideakitai/ArtNet #47: Issues using the ArtnetSender

johnHouse82, 2023-08-01:
Hi hideakitai,

I'm having some problems with getting the Sender to work but I think its more my comprehension than anything else.  I'm looking to send commands to Universe 0, DMX channel 1 which I want to increase the value to 255 or reduce to 0 respectively based on an If Statement.  I am using a Arduino Uno and an Ethernet shield  I have a few questions;

//setup Ethernet I'm guessing is setting up some other function independent from your library? This wold require me to include something else to manage the Ethernet?

I'm guessing that I want to send a "one line send" but I'm not sure what the list of parameters on the 'artnet.send' command refer to (("127.0.0.1", universe15bit, data_ptr, size);)

Thanks for any help or direction
John

hideakitai, 2023-08-01:
Please take a look at an example. Everything you want is there. data_ptr and size are the same ones used there.

https://github.com/hideakitai/ArtNet/blob/master/examples/Ethernet/sender/sender.ino

johnHouse82, 2023-08-02:
Thanks for the pointer, this looks a lot more clear than the cover page :)

John

hideakitai, 2023-08-02:
If you still have a problem, please reopen this issue.

closed 2023-08-02 by hideakitai

johnHouse82, 2023-08-02:
So, I've tried to splice it into my code which I'm working on and I've still got a few questions which I've left in the comments if you have a few mins to glance at them.  I'm aware that this currently won't run due to the multiple calls running at once but if you happen on any suggestions regarding the project they will be appreciated.

John

FYI Project Goal - Big Red Button that when HIGH stops a single value on the desk and when LOW reactivates it

`//Always comment your code like it will be maintained by a violent psychopath who knows where you live
// This code will detect whether or not the Emergency Stop button is depressed.  Aww, so sad ;)
// V2.x is intended for use over Artnet.
//Libraries
#include <ArtnetEther.h>

//Constants
// Ethernet settings
const IPAddress ip(192, 168, 0, 201);                           //set Arduinos IP Address
uint8_t mac[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB};           //set Arduinos MAC Address?

ArtnetSender artnet;                                            //Create ArtNet sender object
const String target_ip = "192.168.0.200";                       //IP address of desk - change to correct address
uint32_t universe = 1;                                          //Target Universe for ArtNet command

const uint16_t size = 512;                                      //Number of channels in the universe? New value of DMX setting?
uint8_t data[size];                                             //???
uint8_t value = 0;                                              //???

int buttonPin = 12;                                             //Declare Pin D12 on the Arduino as an object
int buttonState = 0;                                  //Define a Varible for tracking the button state
int lastButtonState = 0;                              //Define a Variable to recall the previous state of the button

//Setup Loop
void setup() {
    Serial.begin(9600);                                         //changed for Baud rate matching computer's serial monitor
    delay(2000);                                                //two second delay - why is this in here?

    Ethernet.begin(mac, ip);                                    //Initialise the Ethernet, setting MAC address and IP
    artnet.begin();                                             //Initialise the ArtNet object
    // artnet.begin(net, subnet);                               //**optionally you can change**  Set Netmask and Subnet gateway

    Serial.println("start");                                    //printout "start"
    Serial.println(UDP_TX_PACKET_MAX_SIZE);                     //printout something, not sure what or why

    pinMode(buttonPin, INPUT_PULLUP);                   //Specifies the interactions of the defined buttonPin to an input GPIO (Estop)
}

//main loop
void loop() {
  value = (millis() / 4) % 256;                               //Revalue the "value" variable.  What is the purpose?
  memset(data, value, size);                                  //Not sure of the funtion or Purpose of this memeset, guessing that its the DMX request on a single request?

  artnet.streaming_data(data, size);                          //Not sure what the rest of this does either, guessing that its the DMX request on consistant
  artnet.streaming(target_ip, universe);                      // automatically send set data in 40fps
    // artnet.streaming(target_ip, net, subnet, univ);          // or you can set net, subnet, and universe

  //  There is a current confliction with the Arduino being asked to perform multiple tasks
  //  with the above code conflicting with the below if statement

  buttonState = digitalRead(buttonPin);               //Read pin D12
  delay(50);                                        //Delay a little bit to avoid bouncing

  if (buttonState != lastButtonState) {               //If the button state is the same as previous button state
    if (buttonState == HIGH) {                        //if the button is not despressed...  
      //ArtNet message for restart goes here
      Serial.println("Message 1 sent");
      delay(1000);                                    //Delay between sending messages
    } else {                                          //if it is not...
      //ArtNet message for shutdown goes here
      Serial.println("Message 2 sent");
      delay(1000);            //Delay between sending messages                                     // A one second delay in the program before looping back to send the packet again
    }
  }
  lastButtonState = buttonState;                      //Save the current button state as the last state, for next time through the loop
}

//Functions`

johnHouse82, 2023-08-02:
So, I've tried to splice it into my code and I've left a few questions in the comments that I'm still confused about if you have the time to take a glance.  I know that the complete project isn't in working order yet as I'll need to work in some sort of state machine to get it to work, but if you have any pointers of suggestions whilst you looking they will be apprieciated

John

FYI - Project Goal is a big red button that can either stop a value on the desk when the button is HIGH or restart the value if LOW.  ArtNet so that whereever it is plugged into the matching network it contacts the desk

```
//Always comment your code like it will be maintained by a violent psychopath who knows where you live
// This code will detect whether or not the Emergency Stop button is depressed.  Aww, so sad ;)
// V2.x is intended for use over Artnet.
//Libraries
#include <ArtnetEther.h>

//Constants
// Ethernet settings
const IPAddress ip(192, 168, 0, 201);                           //set Arduinos IP Address
uint8_t mac[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB};           //set Arduinos MAC Address?

ArtnetSender artnet;                                            //Create ArtNet sender object
const String target_ip = "192.168.0.200";                       //IP address of desk - change to correct address
uint32_t universe = 1;                                          //Target Universe for ArtNet command

const uint16_t size = 512;                                      //Number of channels in the universe? New value of DMX setting?
uint8_t data[size];                                             //???
uint8_t value = 0;                                              //???

int buttonPin = 12;                                             //Declare Pin D12 on the Arduino as an object
int buttonState = 0;                                  //Define a Varible for tracking the button state
int lastButtonState = 0;                              //Define a Variable to recall the previous state of the button

//Setup Loop
void setup() {
    Serial.begin(9600);                                         //changed for Baud rate matching computer's serial monitor
    delay(2000);                                                //two second delay - why is this in here?

    Ethernet.begin(mac, ip);                                    //Initialise the Ethernet, setting MAC address and IP
    artnet.begin();                                             //Initialise the ArtNet object
    // artnet.begin(net, subnet);                               //**optionally you can change**  Set Netmask and Subnet gateway

    Serial.println("start");                                    //printout "start"
    Serial.println(UDP_TX_PACKET_MAX_SIZE);                     //printout something, not sure what or why

    pinMode(buttonPin, INPUT_PULLUP);                   //Specifies the interactions of the defined buttonPin to an input GPIO (Estop)
}

//main loop
void loop() {
  value = (millis() / 4) % 256;                               //Revalue the "value" variable.  What is the purpose?
  memset(data, value, size);                                  //Not sure of the funtion or Purpose of this memeset, guessing that its the DMX request on a single request?

  artnet.streaming_data(data, size);                          //Not sure what the rest of this does either, guessing that its the DMX request on consistant
  artnet.streaming(target_ip, universe);                      // automatically send set data in 40fps
    // artnet.streaming(target_ip, net, subnet, univ);          // or you can set net, subnet, and universe

  //  There is a current confliction with the Arduino being asked to perform multiple tasks
  //  with the above code conflicting with the below if statement

  buttonState = digitalRead(buttonPin);               //Read pin D12
  delay(50);                                        //Delay a little bit to avoid bouncing

  if (buttonState != lastButtonState) {               //If the button state is the same as previous button state
    if (buttonState == HIGH) {                        //if the button is not despressed...  
      //ArtNet message for restart goes here
      Serial.println("Message 1 sent");
      delay(1000);                                    //Delay between sending messages
    } else {                                          //if it is not...
      //ArtNet message for shutdown goes here
      Serial.println("Message 2 sent");
      delay(1000);            //Delay between sending messages                                     // A one second delay in the program before looping back to send the packet again
    }
  }
  lastButtonState = buttonState;                      //Save the current button state as the last state, for next time through the loop
}

//Functions
```