hideakitai/ArtNet #53: Why can't Mega2560 control artnet signals? I have successfully controlled using Arduino Uno!!

Axx2017, 2023-08-16:
Why do I use mega2560 ss53, Ethernet. init (53); It can be accessed through IP, such as 192.168.1.111. However, cannot artnet data be sent to the art controller? I have successfully controlled the program using Arduino Uno, only adding Ethernet. init (53) in begin();.

Axx2017, 2023-08-16:
@hideakitai c

Axx2017, 2023-08-16:
@hideakitai 
I saw an update to the library, but it doesn't seem to have resolved this issue. Could you please help me answer or solve this problem?

hideakitai, 2023-08-17:
Sorry, I couldn’t understand what `Ethernet.init (53)` means. Please try examples as is (you may need to change up at least)

hideakitai, 2023-08-24:
Closed due to lack of information. Please reopen if you still have problems.

closed 2023-08-24 by hideakitai