hideakitai/ArtNet #55: Can't make example files work

AllanCLight, 2023-09-08:
Hi, i tried to used two of your example files ( artnet receiver & artnet fastled ) and i've got some issue. 

I used them on arduino Uno, then on Mega because i read about lack of memory from the uno.

I can ping my device form my PC so the connection seems to be ok, the orange Led is blinking and when i send Artnet from Luminet Monitor Software it goes straight orange so i think the arduino receive the artnet packet but nothing in the serial monitor when using artnet receiver and no control over my LED when using artnet fast led ( ofc when i used my LED only with fastled program it works, any hardware trouble). 

Have u got any idea ? :) 

Thanks a lot for this great library

hideakitai, 2023-09-20:
Hmm... please check

- firewall (if you use Windows)
- universe config (matched?)
- dmx channel (on your software)

Anyway, some configuration mismatch may cause your error. Please let me know the detailed environment information if you still need help.

hideakitai, 2023-09-25:
If you still need help, feel free to reopen this issue.

closed 2023-09-25 by hideakitai

AllanCLight, 2023-12-14:
Hi, thx for the answer ! I work back on this project and i found some answer on forum. I didn't know that Ethernet Shield can't work when pluged directly on my pc Ethernet Port.  It need to path through a router. So i tried with the ChatServeur and i works !! Then i tried with your sender exemple and my Artnet Monitor Software received packet from arduino, great ! But i can't see anything in the serial monitor when i use the receive exemple sketch.. Does your library read Broadcast packet or does they need to be unicasted ? Caus i suspect my software only to unicast packet..