hideakitai/ArtNet #59: configuring a setup with more than 4 universes

cyberboy666, 2023-09-13:
im wanting to control an installation of probably 1000 leds or more. for this will need to be listening across more than 4 universes - i came up first against the limit of 4 callbacks - wondering why this limit is here? 

is there a better way than using callbacks to subscribe to this many universes ?

(i tried bypassing this in the code and it still compiled and uploaded but im guessing theres a reason not to do it like this lol)

hideakitai, 2023-09-20:
The limitation of the number of callbacks is based on the PollReply packet. PortTypes, GoodInput/Output, SwIn, etc., are limited to 4 ports. But it is better to subscribe to more than four callbacks and make public 4 of them. So, I will remove the limitation of callbacks. :)

hideakitai, 2023-09-21:
@cyberboy666 Fixed. Please check it out

closed 2023-09-21 by hideakitai

cyberboy666, 2023-09-28:
thank you very much for quick response and updates :heart: :heart: