hideakitai/ArtNet #61: ArtnetETH version not working

kilrah, 2023-09-24:
2 issues it seems:
- Missing `#include <EthernetUdp.h>` in `ArtnetEth.h`
- The template substitution fails for `localIP()`, `localIP()` and `macAddress()` in ArtnetCommon.h:519-521

```
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h: In instantiation of 'void arx::artnet::Receiver_<S>::poll_reply() [with S = WiFiUDP]':
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:320:25:   required from 'arx::artnet::OpCode arx::artnet::Receiver_<S>::parse() [with S = WiFiUDP]'
C:\Users\<redacted>\<redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:173:18:   required from here
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:519:23: error: no matching function for call to 'arx::artnet::Receiver_<WiFiUDP>::localIP()'
             IPAddress my_ip = localIP();
                       ^~~~~
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted><redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:614:14: note: candidate: 'template<class T> arx::stdx::enable_if_t<std::is_same<T, EthernetUDP>::value, IPAddress> arx::artnet::Receiver_<S>::localIP() [with T = T; S = WiFiUDP]'
         auto localIP() -> std::enable_if_t<std::is_same<T, EthernetUDP>::value, IPAddress> {
              ^~~~~~~
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:614:14: note:   template argument deduction/substitution failed:
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted><redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:520:23: error: no matching function for call to 'arx::artnet::Receiver_<WiFiUDP>::subnetMask()'
             IPAddress my_subnet = subnetMask();
                       ^~~~~~~~~
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted><redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:618:14: note: candidate: 'template<class T> arx::stdx::enable_if_t<std::is_same<T, EthernetUDP>::value, IPAddress> arx::artnet::Receiver_<S>::subnetMask() [with T = T; S = WiFiUDP]'
         auto subnetMask() -> std::enable_if_t<std::is_same<T, EthernetUDP>::value, IPAddress> {
              ^~~~~~~~~~
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:618:14: note:   template argument deduction/substitution failed:
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted><redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:521:13: error: no matching function for call to 'arx::artnet::Receiver_<WiFiUDP>::macAddress(uint8_t [6])'
             macAddress(r.mac);
             ^~~~~~~~~~
In file included from c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/ArtnetETH.h:13,
                 from C:\Users\<redacted><redacted>\Swing Nozzle\artnet_stepper_control\artnet_stepper_control.ino:5:
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:622:14: note: candidate: 'template<class T> arx::stdx::enable_if_t<std::is_same<T, EthernetUDP>::value> arx::artnet::Receiver_<S>::macAddress(uint8_t*) [with T = T; S = WiFiUDP]'
         auto macAddress(uint8_t* mac) -> std::enable_if_t<std::is_same<T, EthernetUDP>::value> {
              ^~~~~~~~~~
c:\Users\<redacted>\Documents\Arduino\libraries\ArtNet/Artnet/ArtnetCommon.h:622:14: note:   template argument deduction/substitution failed:
```

hideakitai, 2023-09-25:
Thank you @kilrah ! Fixed by #62 

Please check if it works in your environment.

closed 2023-09-25 by hideakitai

kilrah, 2023-09-25:
Oof I should have found this one!
Yep works fine, thanks!