hideakitai/ArtNet #72: receive_fastled on esp32 platform io vscode return WiFi.softAPSubnetMask();

ririfonfon, 2023-10-24:
hello,
i try to compile receive_fastled.
i have this erreur :
`.pio/libdeps/esp32dev/ArtNet/Artnet/ArtnetCommon.h:590:29: error: 'class WiFiClass' has no member named 'softAPSubnetMask'; did you mean 'softAPSubnetCIDR'?`

of course because i don't have softAPSubnetMask in WiFiAP.cpp

in line 590 ArtnetCommon.h
i change ` return WiFi.softAPSubnetMask();
to return WiFi.softAPSubnetCIDR(); `

but reeboot when the esp recieve artnet

ririfonfon, 2023-11-02:
hello,
i change the version 0.2.12
 `lib_deps = 
	hideakitai/ArtNet @ ~0.2.12`

but now i have this erreur:
`.pio/libdeps/esp32dev/ArtNet/Artnet/ArtnetCommon.h:385:21: error: no matching function for call to 'std::map<unsigned int, std::function<void(const unsigned char*, short unsigned int)> >::insert(std::pair<unsigned int, std::function<void(unsigned char*, short unsigned int)> >)'`

ririfonfon, 2023-11-02:
ok 
i try 
`lib_deps = hideakitai/ArtNet`

`platform = espressif32`
`platform = espressif32 @ ^6.0.1`
`platform = https://github.com/platformio/platform-espressif32.git`

the same erreur to all platform:
` .pio/libdeps/esp32dev/ArtNet/Artnet/ArtnetCommon.h:399:17: error: no matching function for call to 'std::map<unsigned int, std::function<void(const unsigned char*, short unsigned int)> >::insert(std::pair<unsigned int, std::function<void(unsigned char*, short unsigned int)> >)'`

hideakitai, 2023-11-02:
Could you show me the whole platmorom.ini file?

ririfonfon, 2023-11-02:
`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
; platform = espressif32 @ ^6.0.1
; platform = https://github.com/platformio/platform-espressif32
; platform = https://github.com/platformio/platform-espressif32.git
; platform = https://github.com/platformio/platform-espressif32#v3.2.0 
board = esp-wrover-kit
framework = arduino
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_speed = 115200
board_build.partitions = partitions.csv
lib_deps = 
	hideakitai/ArtNet
	; hideakitai/ArtNet @ ~0.2.12`

ririfonfon, 2023-11-06:
ok I found it,
It was the LED pointers that were bad

closed 2023-11-06 by ririfonfon