hideakitai/ArtNet #73: How to send an Artnet Trigger / Custom OpCodes

itsyoshio, 2024-01-03:
Hi,

i'd like to send (or stream) packages with different OpCodes.
Specifically i want to send an ArtNetTrigger (0x9900) to a Mixer.

I've found changing 

```c
packet[IDX(Index::OP_CODE_H)] = (OPC(OpCode::Dmx) >> 8) & 0x00FF;
packet[IDX(Index::OP_CODE_L)] = (OPC(OpCode::Dmx) >> 0) & 0x00FF;
```

to 

```c
packet[IDX(Index::OP_CODE_H)] = (OPC(OpCode::Trigger) >> 8) & 0x00FF;
packet[IDX(Index::OP_CODE_L)] = (OPC(OpCode::Trigger) >> 0) & 0x00FF;
```

from [ArtnetCommon.h#L218](https://github.com/hideakitai/ArtNet/blob/f0cd567fb36f76d74077056e44181ab31b2af5e7/Artnet/ArtnetCommon.h#L218)

Works, but is there a better way to set the OpCode?

<details>
  <summary>This is the Code i'm using right now</summary>
  
```c
#define ARTNET_TRIGGER 9900
struct ArtNetTriggerPacket {
    uint8_t ID[18];
    uint16_t OpCode;
    uint8_t ProtVerHi;
    uint8_t ProtVerLo;
    uint8_t Sequence;
    uint8_t Physical;
    uint8_t SubUni;
    uint8_t Net;
    uint8_t Command;
    uint8_t Key[512];
};

// i've tried 18 because that's the Index of OpCode::Trigger in ArtnetCommon.h

// Create an instance of the packet
ArtNetTriggerPacket packet;

// [...]

// memcpy(packet.ID, "Art-Net\0", 8);
// packet.OpCode = htons(ARTNET_TRIGGER);
// packet.ProtVerHi = 0;
// packet.ProtVerLo = 14;
// packet.Sequence = 0;
// packet.Physical = 0;
// packet.SubUni = 0;
// packet.Net = 0;
// packet.Command = d.toInt(); // Fill this in with the appropriate command
// memset(packet.Key, 0, sizeof(packet.Key));
// artnet.send(target_ip, universe, (uint8_t*)&packet, sizeof(packet)); // one-line send
memcpy(packet.ID, d.c_str(), 18);
artnet.send(target_ip, universe, (uint8_t*)&packet, sizeof(packet)); // one-line send
```

The Package this sends (UNPATCHED) looks like this in WireShark :
[image]

</details>

hideakitai, 2024-01-04:
Thank you for reporting. It's good to add an API to send custom packets. I've added the `send_raw()` method to send your packet without modifying the library code. 

https://github.com/hideakitai/ArtNet/pull/74

I may add a feature to send `ArtTrigger` later if I have time :)

hideakitai, 2024-01-04:
ArtTrigger feature was merged. Could you try if it works?

https://github.com/hideakitai/ArtNet/pull/76/files

- configure by `set_oem()` `set_key()` `set_subkey()` `set_payload()`
- send trigger by `trigger()`

hideakitai, 2024-01-04:
Thanks! Fixed https://github.com/hideakitai/ArtNet/pull/78

Could you try it once again?

itsyoshio, 2024-01-05:
> Thanks! Fixed #78
> 
> Could you try it once again?

Thank you!
The Payload is now working perfectly. The Subkey still seems to be 0xff always.
[image]

<details>
  <summary>Code</summary>
  
```c
artnet.set_oem(0x7df1);
artnet.set_oem(0xfd); // subkey
artnet.set_key(0x00);
artnet.set_payload((uint8_t*)d.c_str(), d.length());
artnet.trigger(target_ip);
```
  
</details>

Thank you so much for your help! 🎉

hideakitai, 2024-01-05:
I wonder if you call set_oem() twice instead of set_subkey()?

itsyoshio, 2024-01-05:
Yes, that's it. Thank you. It's all working perfectly.

I was looking at [this Line](https://github.com/hideakitai/ArtNet/pull/76/commits/c5f834b2bc10d0c39226b6c48534b8d4e3ff782f#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R272) and mistakenly thought that 
- set_oem with 16 Bit => OEM Code
- set_oem with 8 Bit   => SubKey

[image]

<details>
<summary>Full Example</summary>

```c
#include <ArtnetETH.h>

// Ethernet stuff
const IPAddress ip(192, 168, 0, 201);
const IPAddress gateway(192, 168, 0, 1);
const IPAddress subnet(255, 255, 255, 0);

ArtnetSender artnet;
const String target_ip = "192.168.0.200";

const uint16_t size = 512;
uint8_t data[size];
uint8_t value = 0;

void setup() {
    Serial.begin(115200);
    delay(2000);

    ETH.begin();
    // ETH.config(ip, gateway, subnet);
    artnet.begin();
    // artnet.begin(net, subnet); // optionally you can change

    artnet.set_oem(0x0f0f);
    artnet.set_subkey(0x0f); // subkey
    artnet.set_key(0x00);

    Serial.println("start");
}

void loop() {
    value = (millis() / 4) % 256;
    memset(data, value, size);

    artnet.set_payload(data, size);
    artnet.trigger(target_ip);
    delay(500);
}

```

</details>

<details>
<summary>Full Example for wESP32</summary>

```c
#include <Arduino.h>
#include <ETH.h>
#include <ArtnetETH.h>

ArtnetSender artnet;
const String target_ip = "192.168.0.200";

const uint16_t size = 512;
uint8_t data[size];
uint8_t value = 0;

// Ethernet stuff
const IPAddress ip(192, 168, 0, 201);
const IPAddress gateway(192, 168, 0, 1);
const IPAddress subnet(255, 255, 255, 0);

static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data) {
  uint8_t mac_addr[6] = {0};
  /* we can get the ethernet driver handle from event data */
  esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

  switch (event_id) {
  case ETHERNET_EVENT_CONNECTED:
    esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
    #ifdef DEBUG
      Serial.println("Ethernet Link Up");
    #endif
    // Wait for the module to get an IP address
    while (ETH.localIP() == IPAddress(0,0,0,0)) {
      delay(500);
    }
    Serial.print("IP address: ");
    Serial.println(ETH.localIP());
    // Serial.println("Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
    //             mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3],
    //             mac_addr[4], mac_addr[5]);
    break;
  case ETHERNET_EVENT_DISCONNECTED:
    #ifdef DEBUG
      Serial.println("Ethernet Link Down");
    #endif
    break;
  case ETHERNET_EVENT_START:
    #ifdef DEBUG
      Serial.println("Ethernet Started");
    #endif
    break;
  case ETHERNET_EVENT_STOP:
    #ifdef DEBUG
      Serial.println("Ethernet Stopped");
    #endif
    break;
  default:
    break;
  }
}

void setup() {
  Serial.begin(115200);
  delay(2000);

  esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL);
  delay(50);

  ETH.begin(0, -1, 16, 17, ETH_PHY_RTL8201);
  // ETH.config(ip, gateway, subnet);

  artnet.begin();
  // artnet.begin(net, subnet); // optionally you can change

  artnet.set_oem(0x0f0f);
  artnet.set_subkey(0x0f); // subkey
  artnet.set_key(0x00);

  Serial.println("start");
}

void loop() {
    value = (millis() / 4) % 256;
    memset(data, value, size);

    artnet.set_payload(data, size);
    artnet.trigger(target_ip);
    delay(500);
}

```

</details>

Tested on wESP32.
These will send a (millis() / 4) mod 256 Payload every 0.5 Seconds like in your [sender.ino](https://github.com/hideakitai/ArtNet/blob/main/examples/ETH/sender/sender.ino)

Thank you @hideakitai !

closed 2024-01-05 by itsyoshio

hideakitai, 2024-01-05:
どういたしまして！