hideakitai/ArtNet #80: Allow to use ENC28J60 Ethernet module

MrFrangipane, 2024-01-05:
Hi,

Thank you for that owesome library

I need to use a ENC28J60 Ethernet module (with the raspberry pico RP2040)

For now, I manually created a `ArtnetEtherENC.h` file in my library folder.

I have made a pull request #79 as an example

All the best,
Val

MrFrangipane, 2024-01-05:
I forgot to mention something

I also had to change `ArtnetCommon.h` at `624:22` so that it doesn"t error
`'class UIPEthernetClass' has no member named 'MACAddress'`

I'll update the pull request accordingly

hideakitai, 2024-01-05:
@MrFrangipane Thank you for your contribution! Could you add examples and workflows for them?

closed 2024-01-10 by MrFrangipane

albinotoni, 2024-08-22:
Hi!

I seem not to be able to get ArtnetEtherENC.h working with nano and ENC28J60. For me the problem seems to be, that Ethernet.begin() does not connect, but silently fails..

On other projects, I manage by using UIPEthernet.h library.