hideakitai/ArtNet #96: Wrong PollReply

linxlabs, 2024-02-15:
Receiver universes shows in DMX Workshop as DMX Inputs, but must be an DMX Output

hideakitai, 2024-02-15:
Thanks, how about v0.3.0? Can you make a pull request?

hideakitai, 2024-02-15:
Also, I think this is correct, but is it not?

> Receiver universes shows in DMX Workshop as DMX Inputs

linxlabs, 2024-02-15:
Can't check it on 0.3.0.

Receiver code get data from ArtNet and puts it outward. This is an output.
Don't checked this on the sender side yet. Your Sender code gets data from somewhere and puts it to ArtNet. This is an input.

hideakitai, 2024-02-15:
> Can't check it on 0.3.0.

？

> Receiver code get data from ArtNet and puts it outward. This is an output. Don't checked this on the sender side yet. 

No. Receiver does not output dmx data.

> Your Sender code gets data from somewhere and puts it to ArtNet. This is an input.

No. Sender does not received data.

linxlabs, 2024-02-15:
Maybe i'm wrong, but look at your fastled example.

It take data from artnet and put it to leds. In terms of artnet it is an output.

hideakitai, 2024-02-15:
> It take data from artnet and put it to leds. In terms of artnet it is an output.

Receiver receives dmx data via artnet. LEDs are outside of artnet. So it's input.

> Seems that sender don't handle ArtPoll at all

Yes. Sender does not receive artnet packet. Just send artdmx data. So it's output.

hideakitai, 2024-02-15:
If you still need help, please feel free to reopen this issue.

closed 2024-02-15 by hideakitai