hideakitai/ArtNet #97: ArtPoolReply

alexbst, 2024-02-17:
ArtPollReplay return wrong universes numbers.

I use subscribeArtDmxUniverse and subscribe 2 universes no matter what universes subscribe to receive ArtPollReplay return 2 times 1

hideakitai, 2024-02-17:
ArtPollReply will be sent once per one subscribed universe now. See below

https://github.com/hideakitai/ArtNet/issues/81

alexbst, 2024-02-18:
Yes
But i subscribe universes 2 and 3 , it 2 times return 1 .

hideakitai, 2024-02-18:
Could you give me reproducible, compilable, and minimum code?

hideakitai, 2024-02-19:
I've tested v0.4.4 https://github.com/hideakitai/ArtNet/blob/main/examples/WiFi/receiver/receiver.ino 

This example correctly sent universes 1 and 2. Also, changing universes to 2 and 3 correctly sends 2 and 3.

hideakitai, 2024-02-28:
If you still need help, please feel free to reopen this issue.

closed 2024-02-28 by hideakitai