hideakitai/ArtNet #98: How to start?

AdvdLaar, 2024-02-22:
Hello,

I want to use this artnet for my home automation (home assistant). I am a hardware engineer, so normally I know what I am doing. Sometimes schematics are provided and a simple example. But none of this. I presume this artnet software uses a RS485 transceiver to communicate with the DMX lights. Which UART can I use? I wanted to use this Artnet wifi project (https://github.com/mtongnz/ESP8266_ArtNetNode_v2), but it uses some outdated JSON bull shit. So no luck eather. Does this Artnet release use a RX and TX from a UART? Do you also use the direction bit? Any directions how to start with this would be great. My device will be controlled by home assistant, sends data to DMX302 by using an ESP8266 and a MAX485.

Best regards,
Ad

hideakitai, 2024-02-22:
This library is an Artnet communication library, and you are free to connect any kind of LEDs beyond it with any kind of hardware. You can connect them as you like.

hideakitai, 2024-02-28:
If you still need help, please feel free to reopen this issue.

closed 2024-02-28 by hideakitai