hideakitai/ArtNet #99: ESP8266 (d1 mini clone) crashes using artnet.forwardArtDmxDataToFastLED

Themicles, 2024-05-09:
Crash occurs right after the following serial message:
WARN: ArtNet packet size is less than requested LED numbers to forward
      requested: 19362      received : 512

That number after requested: doesn't change no matter how many LEDs I set in NUM_LEDS

I cannot make the loop version from the receive_fastled example work either, but there's no crash.

However I am able to use the receive example with the serial readout of the DMX values and that works fine.

hideakitai, 2024-05-09:
Could you give me a minimal code to reproduce your problem in my environment?

Themicles, 2024-05-09:
The following code is just a cut down version of the example receive_fastled and results in the same crash
WARN: ArtNet packet size is less than requested LED numbers to forward
      requested: 32286      received : 512

Note the requested number is this time different than my previous attempts.

```
#include <FastLED.h>  // include FastLED *before* Artnet

// Please include ArtnetWiFi.h to use Artnet on the platform
// which can use both WiFi and Ethernet
#include <ArtnetWiFi.h>
// this is also valid for other platforms which can use only WiFi
// #include <Artnet.h>

// WiFi stuff
const char* ssid = "your-ssid";
const char* pwd = "your-password";

ArtnetWiFiReceiver artnet;
uint8_t universe = 1;  // 0 - 15

// FastLED
#define NUM_LEDS 2
CRGB leds[NUM_LEDS];
const uint8_t PIN_LED_DATA = 12;

void setup() {
    Serial.begin(115200);
    delay(2000);

    FastLED.addLeds<NEOPIXEL, PIN_LED_DATA>(leds, NUM_LEDS);

    // WiFi stuff
    WiFi.begin(ssid, pwd);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.print("WiFi connected, IP = ");
    Serial.println(WiFi.localIP());

    artnet.begin();

    // if Artnet packet comes to this universe, forward them to fastled directly
    artnet.forwardArtDmxDataToFastLED(universe, leds, NUM_LEDS);
}

void loop() {
    artnet.parse();  // check if artnet packet has come and execute callback
    FastLED.show();
}
```

Themicles, 2024-05-09:
Taking a stab at debugging this myself, I've narrowed it down to var "num" from 
`void forwardArtDmxDataToFastLED`
does not survive past
`this->subscribeArtDmxUniverse(universe, [&](const uint8_t* data, const uint16_t size, const ArtDmxMetadata &metadata, const RemoteInfo &remote) {`
I came to this by doing a Serial.print of num at various points in the code until it stopped report "2"

Despite more than a decade of debugging code, I still consider myself a hobbyist and don't really understand why that would be the case.

hideakitai, 2024-05-19:
Thanks, https://github.com/hideakitai/ArtNet/pull/100 will fix it

hideakitai, 2024-05-19:
@Themicles [v0.4.5](https://github.com/hideakitai/ArtNet/releases/tag/v0.4.5) released. Please try it

Themicles, 2024-05-20:
Sorry about the delayed response, I was on site working on some prep for a show next weekend.

I have updated the library and recompiled, then uploaded the sketch to the ESP8266. Had to switch computers because the upload kept getting garbled, but after doing so, it is working as expected. Thanks!

closed 2024-05-20 by Themicles