hideakitai/ArtNet #105: trying to set up Artnet for both Wifi & Ethernet not working

markrosa, 2024-06-13:
I have had success using ArtnetWifi and ArtnetEther on my PicoW using a WIZnet Ethernet Hat (W5100S).
Now I wanted to try to set the system up in a way that it tries to connect via Ethernet first (upon success starting Artnet functionality as a DMX Receiver for two universes) and if there is no Ethernet connection it would try to connect to Wifi with the given credentials (again starting Artnet functionality as a DMX Receiver for two universes) and only if that also fails it would present an error on the small screen attached.

When I follow the recommendations in the Readme.md file, to not include <ArtnetWifi.h> or <ArtnetEther.h> but instead use <Artnet.h>, I get the error "BOTH WIFI AND ETHERNET ARE ENABLED. PLEASE INCLUDE ONLY ArtnetWifi.h or ArtnetEther.h".
Is that expected behaviour?

hideakitai, 2024-06-13:
Yes, you can use either WiFi or Ethernet

markrosa, 2024-06-14:
Thank you for your quick response, but unfortunately this does not answer my question:
Why does it show this error?
So, is there no way to use both Wifi and Ethernet (at least to initialize both) and then check for the two types of connections one after the other and use the one that works first?

hideakitai, 2024-06-14:
It is not impossible to do so, but various conflicts arise when using them simultaneously. Also, there are many different types of WiFi and Ethernet (WiFiNINA, WiFiS3, ETH, EthernetENC, etc.), and it takes a lot of effort to make them all work without problems. Since this library is something I maintain in my spare time, it is not currently supported due to the balance between allocating that effort and user requests.

If you could offer it to me as a job, I would try to set aside some time to deal with it.

markrosa, 2024-06-14:
Thanks for the explanation.
All fine, I will try to find a solution around it.

But maybe you should take out the hint then about including <Artnet.h> instead of  <ArtnetWifi.h> or <ArtnetEther.h> for platforms that provide both Wifi and Ethernet capabillities. As this leads to the assumption that both can be used in parallel.

Thanks for your effort spent!

Btw. I am trying to use your software in my free time for usage in an Artnet DMX interface that I am building for the school my kids go to.

hideakitai, 2024-06-14:
No, it is not. For boards that can use both WiFi and Ethernet, such as the ESP32 for example, including ArtNet.h will cause conflicts, so it throws an error before doing so. You have to choose one or the other. To use both at the same time, a major modification of the library is required.

hideakitai, 2024-06-14:
> Btw. I am trying to use your software in my free time for usage in an Artnet DMX interface that I am building for the school my kids go to.

Great, I hope your kids enjoy it ;)

closed 2024-06-15 by hideakitai