hideakitai/ArtNet #107: ArtPollReply-Universe In

dantheman-22, 2024-06-18:
I'm using the latest 0.5.0 and the device is discoverable by Madrix 5 however I do not want it to show Universe In.
In Madrix it shows Universe Out and Universe In as the same (universe 1).  How do I disable the Universe In so this will not show in the broadcast request?  I'm currently passing the reply values like this.
        artnet.begin();
        // Initialize ArtPollReplyConfig
        madrix.short_name = deviceNameString;
        madrix.long_name = "Winch Board";
[image]

        // Set ArtPoll Reply Config
        artnet.setArtPollReplyConfig(
            madrix.oem,
            madrix.esta_man,
            madrix.status1,
            madrix.status2,
            madrix.short_name,
            madrix.long_name,
            madrix.node_report
        );

hideakitai, 2024-06-18:
If you subscribe to any universe, that universe is automatically set. If you use subscribeArtDmx() instead of subscribeArtDmxUniverse(), universe in will be set to zero.

dantheman-22, 2024-06-18:
Thanks for the reply and the information hideakitai!  I need it to listen to the universe that is specified and not all universes.  Is there a way to subscribeArtDmxUniverse() but have universe in as nothing?  Or is this a library limitation?
I changed this to the below but noticed that I am unable to supply the universe it is suppose to be listening too.

artnet.subscribeArtDmxUniverse(artnetUniverse, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {}

artnet.subscribeArtDmx(artnetUniverse, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {}

hideakitai, 2024-06-18:
You can identify the universe using subscribeArtDmx(). Please see the example for more detail https://github.com/hideakitai/ArtNet/blob/007da61549a697937b6c7c1f5452fbf5f8f8c6f3/examples/WiFi/send_receive/send_receive.ino#L56

dantheman-22, 2024-06-19:
when using artnet.subscribeArtDmx it is not discoverable.  The device does not respond to a broadcast packet with any information.  Is that how it should be?
I'll include my setup if you wouldn't mind advisng me the best way to achieve my goal.  

`artnet.subscribeArtDmxUniverse(artnetUniverse, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
            
            // Ignore Broadcast Packets from any other source if Unicast IP is Set
            if (isUnicastIPSet) {
                if (remote.ip != unicastIP) {
                    return;
                }
            }

            // Calculate the number of effective groups of LEDs
            size_t numGroups = globalnumpixel1 / globalgroup1;

            // Adjust the start index based on the globalstartchan1 value
            size_t startIdx = (globalstartchan1 - 1); // DMX channels are 1-based

            for (size_t group = 0; group < numGroups; ++group) {
                // Calculate the starting index for the current group
                size_t idx = startIdx + group * 3 * globalgroup1;

                // Ensure idx is within bounds
                if (idx + 2 < size) {
                    // Get the color data for the current group
                    uint8_t r = data[idx];
                    uint8_t g = data[idx + 1];
                    uint8_t b = data[idx + 2];

                    // Apply the intensity scaling if needed
                    if (globalIntensity != 255) {
                        r = r * globalIntensity / 255;
                        g = g * globalIntensity / 255;
                        b = b * globalIntensity / 255;
                    }

                    // Update all LEDs in the current group with the same color data
                    for (size_t pixel = 0; pixel < globalgroup1; ++pixel) {
                    size_t ledIndex = group * globalgroup1 + pixel;
                    if (globalOrder == "RGB") {
                        leds[ledIndex].setRGB(r, g, b);
                    } else if (globalOrder == "GRB") {
                        leds[ledIndex].setRGB(g, r, b);
                    } else if (globalOrder == "BRG") {
                        leds[ledIndex].setRGB(b, r, g);
                    } else if (globalOrder == "GBR") {
                        leds[ledIndex].setRGB(g, b, r);
                    } else {
                        leds[ledIndex].setRGB(r, g, b);
                    }
                }
                }
            }
        });`

hideakitai, 2024-06-19:
Thanks, I confirmed it. I'll fix it so that if there is no universe to subscribe to, it sends a poll reply with universe-in as 0.

hideakitai, 2024-06-19:
v0.6.0 fixes the issue. Please feel free to reopen this issue if you still need help.

closed 2024-06-19 by hideakitai