hideakitai/ArtNet #110: Subscribed Universe Not working for any other than 0

dantheman-22, 2024-06-26:
If I subscribe to any other universe other than 0 it does not work.  See below example code I'm using.  Universe 0 works fine but setting to universe 1 won't receive any data.
` uint16_t artnetUniverse = 1
artnet.subscribeArtDmxUniverse(artnetUniverse, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {// Debugging output for received packets
    Serial.print("Received packet from IP: ");
    Serial.println(remote.ip);
    Serial.print("Packet Universe: ");
    Serial.println(metadata.universe);
    Serial.print("Packet Size: ");
    Serial.println(size);}`

hideakitai, 2024-06-26:
What universe do you send to? The example code works at least in Universe 1 and 2.

dantheman-22, 2024-06-26:
Actually trying to receive any universe between 0-15 and pass to fastled. I need to only subscribe to a single universe or at a minimum listen to only the universe the device is configured for.

hideakitai, 2024-06-26:
Please try receiver example first. Does it work with your sender?

dantheman-22, 2024-06-26:
okay, have tested quite a bit this morning and only universe 0 works with both lambda and pre-defined callbacks.  Sender is sending 8 universes of data.  Noticed that the universe IN changes but the Universe Out does not on the ArtPollReplyConfig.  I have Madrix able to see the device but the universe numbers are not changing for the OUT.  It appears as though the device is only wanting to subscribe to Universe 0.  Any other universe and the library does not want to handle any data.  Pasting the exact code I used as per your comment.  You can download MA Dot 2 onPC and setup a simple ArtNet Network and send a single universe of data to your device to test.  Happy to do testing here to iron out this bug if you want to help me out with other things to try?

`#include <config.h>
#include <ArtnetETH.h>

// Ethernet stuff
const IPAddress ip(10, 0, 0, 100);
const IPAddress gateway(10, 0, 0, 1);
const IPAddress subnet_mask(255, 0, 0, 0);

ArtnetReceiver artnet;
uint16_t universe1 = 1; // 0 - 32767
uint8_t net = 0;        // 0 - 127
uint8_t subnet = 0;     // 0 - 15
uint8_t universe2 = 2;  // 0 - 15

void callback(const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
    // you can also use pre-defined callbacks
}

void setup() {
    Serial.begin(115200);

    ETH.begin(ETH_PHY_ADDR, ETH_PHY_POWER, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_TYPE, ETH_CLK_MODE);
    ETH.config(ip, gateway, subnet_mask);
    artnet.begin();

    // if Artnet packet comes to this universe, this function (lambda) is called
    artnet.subscribeArtDmxUniverse(universe1, [&](const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
        Serial.print("lambda : artnet data from ");
        Serial.print(remote.ip);
        Serial.print(":");
        Serial.print(remote.port);
        Serial.print(", universe = ");
        Serial.print(universe1);
        Serial.print(", size = ");
        Serial.print(size);
        Serial.print(") :");
        for (size_t i = 0; i < size; ++i) {
            Serial.print(data[i]);
            Serial.print(",");
        }
        Serial.println();
    });

    // you can also use pre-defined callbacks
    artnet.subscribeArtDmxUniverse(net, subnet, universe2, callback);
}

void loop() {
    artnet.parse();  // check if artnet packet has come and execute callback
}`

hideakitai, 2024-06-27:
Hm, are you using v0.6.0? How about v0.5.0?

dantheman-22, 2024-06-27:
Tested with both 0.6 and 0.5 and both do the same thing.

hideakitai, 2024-06-27:
I have tested the receiver.ino example on my environment, and worked correctly with Universe 1 and 2. Could you confirm the environment/configuration? 

Also, you can add the callback when the packet toward any universe has come:

```cpp
    artnet.subscribeArtDmx([&](const uint8_t *data, uint16_t size, const ArtDmxMetadata& metadata, const ArtNetRemoteInfo& remote) {
        // if Artnet packet comes, this function is called for every universe
        Serial.print("Universe: ");
        Serial.println(metadata.universe);
    });
```

dantheman-22, 2024-06-30:
apologies for delay - I will have time to work on this tonight and advise.  I'll close this now but will comment if I have issues.

dantheman-22, 2024-07-29:
Have completed some testing and found the following.  Not sure if a bug or limitation at the moment.  I would very much appreciate some feedback and knowledge on this.
If I **subscribeArtDmxUniverse** and set my device to listen to universe '2', my sender when scanning for devices see's the device as 'Universe In as 2' and 'Universe Out as 1'.  The device appears to be listening to universe 1 however my device is seeing it as it should be receiving universe 2.  The device in this scenario should be listening for Universe 2.

If I **subscribeArtDmx** the device receives universe 1 correctly however if I add the following code, the device is still only listening for universe 1 and is not listening to the set universe. artnetUniverse = 2.
if (metadata.universe != artnetUniverse){return;}

Adding the callback for any universe, the serial print log only shows Universe: 0 which is universe 1 however I should expect it to show the universe it is receiving however this value does not change.

I have other devices here that are correctly receiving universe 2 and my sender is correctly sending 8 universes of data.
I've tried v0.5.0, v0.6.0 and v0.7.1.

Have you got any ideas or tips in testing this further?

hideakitai, 2024-07-30:
> If I subscribeArtDmxUniverse and set my device to listen to universe '2', my sender when scanning for devices see's the device as 'Universe In as 2' and 'Universe Out as 1'. The device appears to be [...]

Device is subscribing universe 2 == "device is listening" universe 2. First, this is correct. "Device is listening" universe2 == "Universe in as 2" from the device perspective. Next, this is also correct for me. 

>  The device appears to be listening to universe 1 however my device is seeing it as it should be receiving universe 2. The device in this scenario should be listening for Universe 2.

What do you mean by this? Could you explain in more detail?

hideakitai, 2024-07-30:
> If I subscribeArtDmx the device receives universe 1 correctly however if I add the following code, the device is still only listening for universe 1 and is not listening to the set universe. artnetUniverse = 2.
> if (metadata.universe != artnetUniverse){return;}

How did you get `the device is still only listening for universe 1 and is not listening to the set universe. artnetUniverse = 2`? Please give me a minimal code that can reproduce the problem.

hideakitai, 2024-07-30:
> Adding the callback for any universe, the serial print log only shows Universe: 0 which is universe 1 however I should expect it to show the universe it is receiving however this value does not change.

Have you tried this example? If not, please try to insert your `print()` here and try it again.

https://github.com/hideakitai/ArtNet/blob/9a3d20c95887cc0d6f73197b336bc108cfc42e81/examples/WiFi/receiver/receiver.ino#L20-L22

hideakitai, 2024-07-30:
If there are likely to be multiple issues, it would be helpful to split the issues by topic.

dantheman-22, 2024-07-30:
thanks for the comments, I'll split the issues out by topic

closed 2024-07-30 by dantheman-22