hideakitai/ArtNet #112: Memory Problems

boyi01, 2024-06-30:
Hey,
i use your Library with a Arduino nano.
When i use Version 5.0.0 i have 120byts of free memory.

When i use Version 6.0.0 i have -24 and a lot of errors 

The code did not change

```
/*
This example will receive multiple universes via Artnet and control a strip of sk6812RGBW leds via
Adafruit's NeoPixel library: https://github.com/adafruit/Adafruit_NeoPixel
This example may be copied under the terms of the MIT license, see the LICENSE file for details
*/
#include <ArtnetEther.h>
ArtnetReceiver artnet;
#include <Ethernet.h>
#include <EthernetUdp.h>
#include <SPI.h>

#include <NeoPixelBus.h>

// Neopixel settings
const int numLeds = 80; // change for your setup
const int channelsPerLed = 4;
const int numberOfChannels = numLeds * channelsPerLed; // Total number of channels you want to receive (1 led = 4 channels)
const byte dataPin = 2;
NeoPixelBus<NeoRgbwwFeature, NeoWs2812xMethod> strip(numLeds, dataPin);

// Check if we got all universes
uint16_t universe1 = 0; // 0 - 32767
bool sendFrame = 1; 
int previousDataLength = 0;

// Change ip and mac address for your setup
byte mac[] = {0x04, 0xE9, 0xE5, 0x00, 0x69, 0xEC};
IPAddress ip(192, 168, 1, 177);
IPAddress myDns(192, 168, 1, 1);

struct ArtPollReplyMetadata
{
    uint16_t oem {0x00FF};      // OemUnknown https://github.com/tobiasebsen/ArtNode/blob/master/src/Art-NetOemCodes.h
    uint16_t esta_man {0x0000}; // ESTA manufacturer code
    uint8_t status1 {0x00};     // Unknown / Normal
    uint8_t status2 {0x08};     // sACN capable
    String short_name {"Arduino ArtNet"};
    String long_name {"Ardino ArtNet Protocol by hideakitai/ArtNet"};
    String node_report {""};
};

void setup()
{   
  Serial.begin(115200);
  Ethernet.begin(mac, ip);
  
  artnet.begin();
  
  strip.Begin();
  strip.Show();
  artnet.subscribeArtDmxUniverse(universe1, [&](const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
      // if Artnet packet comes to this universe(0-15), this function is called

      int offset =  1;
        float colorBalance = ((float) data[0])/255;
        // read universe and put into the right part of the display buffer
        for (int i = 0; i < size / channelsPerLed; i++)
        { 
          int led = i;
          if (led < numLeds) {
            strip.SetPixelColor(led, RgbwwColor( data[i * channelsPerLed + offset], data[i * channelsPerLed + 1 + offset], data[i * channelsPerLed + 2 + offset],round(((float) data[i * channelsPerLed + 3 + offset]) * colorBalance),round(((float) data[i * channelsPerLed + 3 + offset]) * abs(colorBalance-1))));
                 //  strip.SetPixelColor(i, RgbwwColor(10,10,10,10,10));
          }

        }
          Serial.println("dmx");
        strip.Show();
 
  });

}
#ifdef __arm__
// should use uinstd.h to define sbrk but Due causes a conflict
extern "C" char* sbrk(int incr);
#else  // __ARM__
extern char *__brkval;
#endif  // __arm__

int freeMemory() {
  char top;
#ifdef __arm__
  return &top - reinterpret_cast<char*>(sbrk(0));
#elif defined(CORE_TEENSY) || (ARDUINO > 103 && ARDUINO != 151)
  return &top - __brkval;
#else  // __arm__
  return __brkval ? &top - __brkval : &top - __malloc_heap_start;
#endif  // __arm__
}
void loop()
{
  Serial.println( freeMemory());
       
  // we call the read function inside the loop
   artnet.parse();
}
```

hideakitai, 2024-06-30:
Arduino Nano doesn't have enough memory. Of course, this library's size will glow if a new feature is added. I recommend using a more powerful board: https://github.com/hideakitai/ArtNet?tab=readme-ov-file#note

boyi01, 2024-06-30:
Okay thanks

Just for being not misunderstood, it's the dynamic part, that means it compiles and even starts :D

I would have expected when I use the same features, memory usage is the same :) 

But still thanks for you work, I use the v5

closed 2024-06-30 by hideakitai