hideakitai/ArtNet #121: ArtPollReply sw_out incorrect

dantheman-22, 2024-07-30:
When the device sends the ArtPollReplyConfig, Lighting Software does not see the SubscribedUniverse correctly.

I have manually edited the library in my setup and confirmed the following fixes the issue.
In ArtPollReply.h on line 109, if I edit the r.sw_out[0] = 0; >  to be > r.sw_out[0] = (universe >> 0); the ArtPollReply Packet sends the correct data for the Lighting Software to broadcast the correct Universe.

With this change, the correct Subscribed Universe is controllable by the software.  The Universe In (r.sw_in) is not really needed but noticed when commenting this out, the library has issues communicating.  Are we able to implement a fix?
See below photo.
[image]

hideakitai, 2024-07-31:
Please give me a minimal code to reproduce the issue.

dantheman-22, 2024-07-31:
`#include <FastLED.h>
#include <ArtnetETH.h>

// Set Artnet Requirements
ArtnetReceiver artnet;

uint16_t globalstartuni1

// Madrix ArtPollReply Config
ArtPollReplyConfig madrix;

void setup() {
preferences.begin("myprefs", false);
globalstartuni1 = preferences.getUInt("start_uni1", 2);
        Serial.print("Start Universe: ");
        Serial.println(globalstartuni1);

FastLED.addLeds<WS2812B, DATA_PIN, RGB>(leds, 10);

uint16_t artnetUniverse = globalstartuni1 - 1;
        Serial.print("Subscribed Art-Net Universe: ");
        Serial.println(artnetUniverse);

//Set ArtPollReply Configuration Data
artnet.setArtPollReplyConfig(
            madrix.oem,
            madrix.esta_man,
            madrix.status1,
            madrix.status2,
            madrix.short_name,
            madrix.long_name,
            madrix.node_report
        );

artnet.subscribeArtDmxUniverse(artnetUniverse, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
}
}`

hideakitai, 2024-07-31:
I couldn't compile it. Please give me a compilable one.

hideakitai, 2024-07-31:
I recommend using code block for the code

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

dantheman-22, 2024-07-31:
your example receive_fastled for the ETH library has the issue but issue is in the ArtPollReply.h file.
```#include <FastLED.h>  // include FastLED *before* Artnet
#include <ArtnetETH.h>

// Ethernet stuff
const IPAddress ip(10, 0, 0, 100);
const IPAddress gateway(0, 0, 0, 0);
const IPAddress subnet(255, 0, 0, 0);

ArtnetReceiver artnet;
uint8_t universe = 2;  // 0 - 15

// FastLED
#define NUM_LEDS 1
CRGB leds[NUM_LEDS];
const uint8_t PIN_LED_DATA = 3;

void setup() {
    Serial.begin(115200);
    delay(2000);

    FastLED.addLeds<NEOPIXEL, PIN_LED_DATA>(leds, NUM_LEDS);

    ETH.begin();
    ETH.config(ip, gateway, subnet);
    artnet.begin();

    artnet.subscribeArtDmxUniverse(universe, [&](const uint8_t* data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
         for (size_t pixel = 0; pixel < NUM_LEDS; ++pixel) {
             size_t idx = pixel * 3;
             leds[pixel].r = data[idx + 0];
             leds[pixel].g = data[idx + 1];
             leds[pixel].b = data[idx + 2];
         }
     });
}

void loop() {
    artnet.parse();
    FastLED.show();
}
```

Here is the line I'm referencing in ArtPollReply.
```
 r.sw_in[0] = (universe >> 0) & 0x0F;
    r.sw_out[0] = (universe >> 0);          // dummy: output port is flexible
```

hideakitai, 2024-07-31:
Thanks, I understand the problem. I should fix 2 problems.

1. As you mentioned, set `sw_out` to subscribed universe
2. Add an API to set `sw_in` manually by the user. Because sending the artnet data from the device is flexible, ArtPollReply cannot dynamically detect what universe is used to send it from the device to the controller. So we should define the `sw_in` in the setup phase by the user manually.

dantheman-22, 2024-07-31:
yea that fix would be perfect - could you include some documentation once you implement this so we know how to utilise the API?

hideakitai, 2024-07-31:
Yes, of course

dantheman-22, 2024-07-31:
thanks for such amazing support on this library.

hideakitai, 2024-07-31:
Please try this branch.

https://github.com/hideakitai/ArtNet/tree/fix/swin-and-swout

You can set swin manually like

```
artnet.setArtPollReplyConfigSwIn(0, 3); // index 0 to universe 3
artnet.setArtPollReplyConfigSwIn(1, 2, 3, 4); // index 0-3 to universe 1, 2, 3, 4
```

dantheman-22, 2024-07-31:
I've tested the new branch and confirm the sw_out is correct now and is correctly showing on the artpollreply.

The sw_in is still showing a '1'.  What am I missing to have it show either '0' or nothing?

Here is the new code in my setup
```
//API as given
artnet.setArtPollReplyConfigSwIn(0, 3);

        // Set ArtPoll Reply Config for Madrix to be discoverable
        artnet.setArtPollReplyConfig(
            madrix.oem,
            madrix.esta_man,
            madrix.status1,
            madrix.status2,
            madrix.short_name,
            madrix.long_name,
            madrix.node_report,
            madrix.sw_in
        );
```

hideakitai, 2024-07-31:
It's great to hear that. I think you can use this.

```
artnet.setArtPollReplyConfigSwIn(1, 2, 3, 4); // sw_in to 1, 2, 3, 4
artnet.setArtPollReplyConfigSwIn(0, 0, 0, 0); // sw_in to 0, 0, 0, 0
```

hideakitai, 2024-07-31:
Or if you want to use `setArtPollReplyConfig()`, 

```c++
        madrix.sw_in[0] = 0;
        madrix.sw_in[1] = 0;
        madrix.sw_in[2] = 0;
        madrix.sw_in[3] = 0;

        artnet.setArtPollReplyConfig(
            madrix.oem,
            madrix.esta_man,
            madrix.status1,
            madrix.status2,
            madrix.short_name,
            madrix.long_name,
            madrix.node_report,
            madrix.sw_in
        );
```

hideakitai, 2024-07-31:
Added API for ease. You can also use this one.

```C++
        madrix.sw_in[0] = 0;
        madrix.sw_in[1] = 0;
        madrix.sw_in[2] = 0;
        madrix.sw_in[3] = 0;

        artnet.setArtPollReplyConfig(madrix);
```

dantheman-22, 2024-07-31:
have tested and it is still responding with '1' for Universe In with the following configuration.  I presume it should be '0' or nothing, it would be nice for it to be '0' at a minimum.
```
ArtPollReplyConfig madrix;

// Initialize ArtPollReplyConfig
        madrix.short_name = deviceNameString;
        madrix.long_name = "Winch Controller";
        //Set Madrix Universe In Per Port - API
        madrix.sw_in[0] = 0;
        madrix.sw_in[1] = 0;
        madrix.sw_in[2] = 0;
        madrix.sw_in[3] = 0;

        artnet.setArtPollReplyConfig(madrix);
```

hideakitai, 2024-07-31:
Does your controller accepts 0 as sw_in? Please paste the controller window.

hideakitai, 2024-07-31:
I think it's a good idea to try setting sw_in to 1, 2, 3, 4, or something to see if it works.

dantheman-22, 2024-08-01:
have tested the following.  set sw_in to 4.  Lighting Software sees it as 5.  Software also supports null values but setting to '0' sets it as '1' in the software.  If you want to test, you can download Madrix 5 for free and just run through the setup to replicate the issue.  Happy to supply further documentation on this if you like?

```
madrix.sw_in[0] = 4;
        madrix.sw_in[1] = 4;
        madrix.sw_in[2] = 4;
        madrix.sw_in[3] = 4;

        artnet.setArtPollReplyConfig(madrix);
```
[image]

dantheman-22, 2024-08-14:
any further update on this one?

hideakitai, 2024-08-14:
No. I don't have the environment to test it now (out on vacation)

hideakitai, 2024-08-14:
I now only have a Mac and a simple ESP32 module, but I noticed that even on a Mac, I can see the ArtPollReply packet itself if I use TouchDesigner, so I gave it a quick try.

I dumped the packets using the `receiver.ino` example as is, which subscribes to universes 1 and 2. You will probably get the following two ArtPollReplies.

```
swin=b'\x00\x00\x00\x00', swout=b'\x01\x00\x00\x00',
```

```
swin=b'\x00\x00\x00\x00', swout=b'\x02\x00\x00\x00', 
```

The result was the same as above; the packets were received as expected. In other words, the contents of both swin and swout packets are as expected.

So, there may be some limitations on the sw_in on the MADRIX side. However, I don't have Windows, so I can't test it now.

You can see the ArtPollReply packets on your end with WireShark or something. Can you investigate? If you run the `receiver.ino` example as is, you will probably receive the same packets.

dantheman-22, 2024-08-14:
I'll do some more wireshark with different configurations and advise the outcome.
Enjoy your vacation :)

dantheman-22, 2024-08-19:
after testing, it appears that the software is requesting a ArtPollReply and is expecting a value to be returned.  I cannot seem to get it to not set it to something so it appears now to be a limitation of the Software and not the library. With the sw_in and sw_out fixes implemented, that has resolved the invalid universe subscription in the software.  
Closing this issue as resolved as the sw_in and sw_out has fixed the original request.

closed 2024-08-19 by dantheman-22

xiooozzz, 2024-11-13:
Let me clear this issue, you cannot use more than one universe in Madrix. No matter how hard you try to tweak ArtPollReply.h. All you can is to change the universe's index. You never simultaneously receive data from two universes. This is a bug in the basic library source-code. I assume when handling the NetSwitch bits in ArtAddress packet, @hideakitai seems locked to use sw_out[0]  to decode a wrong universe, instead of use SwIn[NetSwitch] or SwOut[NetSwitch].

xiooozzz, 2024-11-13:
Let me clear this issue out, by doing this tweak in ArtPollReply.h. You can get what you want (3 port, 3 universe, 3 dmx-out).

`    r.num_ports_h = 0; // Reserved
    r.num_ports_l = 3;
    memset(r.sw_in, 0, 4);
    memset(r.sw_out, 0, 4);
    memset(r.port_types, 0, 4);
    memset(r.good_input, 0, 4);
    memset(r.good_output, 0, 4);
    r.net_sw = (universe >> 8) & 0x7F;
    r.sub_sw = (universe >> 4) & 0x0F;
    // https://github.com/hideakitai/ArtNet/issues/81
    // https://github.com/hideakitai/ArtNet/issues/121
    r.sw_out[0] = 0;
    r.sw_out[1] = 1;
    r.sw_out[2] = 2;
    r.sw_in[0] = 0;
    r.sw_in[1] = 1;
    r.sw_in[2] = 2;
    r.port_types[0] = 0xC0;   // I/O available by DMX512
    r.good_input[0] = 0x80;   // Data received without error
    r.good_output[0] = 0x80;  // Data transmitted without error
    r.port_types[1] = 0xC0;   // I/O available by DMX512
    r.good_input[1] = 0x80;   // Data received without error
    r.good_output[1] = 0x80;  // Data transmitted without error
    r.port_types[2] = 0xC0;   // I/O available by DMX512
    r.good_input[2] = 0x80;   // Data received without error
    r.good_output[2] = 0x80;  // Data transmitted without error
    r.sw_video = 0;   // Video display shows local data
    r.sw_macro = 0;   // No support for macro key inputs
    r.sw_remote = 0;  // No support for remote trigger inputs
    memset(r.spare, 0x00, 3);`