hideakitai/ArtNet #125: RDM?

tonyxforce, 2024-09-01:
Is there chance of this library ever having [RDM](wikipedia.org/wiki/RDM_(lighting)) support? I want to make a DMX interface and i'm modifying [Mtongnz's code](https://github.com/mtongnz/ESP8266_ArtNetNode_v2) for it, but this library seems much more polished and user friendly but it doesn't support RDM.

hideakitai, 2024-10-22:
@tonyxforce Sorry for not getting back to you sooner. I am not familiar with RDM, so I did some quick research. However, there seems to be a lot to fix, so I have plans to support it if many people request it.

tonyxforce, 2024-10-22:
Thank you! Looking forward to it

Setitch, 2025-03-13:
I would love to see that feature

einlichtvogel, 2025-05-28:
would be a +1 here

hideakitai, 2025-06-01:
In response to your request, I have implemented a rough version here. 

- https://github.com/hideakitai/ArtNet/pull/135

It does not yet build on some platforms, but if it builds on your board, please try RDM. Please let me know if you encounter any issues.

einlichtvogel, 2025-06-01:
I tried the example from you with GrandMA3.

Your Libary receives the ArtTodRequest, and sends an TOD response, but that's it.
No other RDM communication.

Do you have an other software where it could be tested?

tonyxforce, 2025-06-01:
Thank you so much for implementing this feature!

The issue I have is that I don't have any RDM enabled Artnet Software OR RDM enabled fixtures

I'm just designing an artnet node in which RDM would be a huge selling point

If you know of any free software that supports RDM please let me know too