hideakitai/ArtNet #126: Sending more than 1 universe?

adeloscope, 2024-09-24:
I have around 500 RGBW pixels I'm controlling and they are organized consecutively to overflow into consecutive universes. The issue I'm having is that when I significantly increase the data size i get some fatal error when I call sendArtDmx(). 

What I don't actually understand is if you do the parsing into universes on the backend? And thus I can send an arbitrary amount of data? Or am I supposed to pre-parse into universe chunks and then call sendArtDmx() however many 512bytes worth of data I have? I was doing that before and was getting very strange results that I couldn't trace. If you could provide working sample code that sent more than one universe that would be amazing.

dantheman-22, 2024-10-04:
Im also chasing some code to send more than one universe to control 240 RGB LEDs which is falling into the second universe but need to control consecutively. 
@adeloscope  have you managed to get more than 170 pixels to work over multiple universes?

dantheman-22, 2024-10-06:
@adeloscope - I've had a bit of time to take a look and managed to get 2 universes to output on the same port.  first universe controls the first 170 pixels (RGB) and the second universe controls the remainder 70 pixels in my scenario.  I have a string of 240 leds and it is working with two universes on one port.  Here is the code I've managed to get to work however, I cannot seem to use a global variable to have the metadata.universe to work.  I have to manually set the universe number in the metadata.universe == 1 as if I add metadata.universe == globaluniverse (which is declared as a uint16_t with a value of 1) it does not work at all.
Maybe you have some ideas on my issue now?
```
void handleUniverse(const uint8_t *data, uint16_t size, size_t startPixel, size_t pixelCount) {
    // Calculate the number of effective groups of LEDs
    size_t numGroups = pixelCount / globalgroup1;
    
    // Adjust the start index based on the globalstartchan1 value
    size_t startIdx = (globalstartchan1 - 1); // DMX channels are 1-based

    for (size_t group = 0; group < numGroups; ++group) {
        // Calculate the starting index for the current group
        size_t idx = startIdx + group * 3 * globalgroup1;
        
        // Ensure idx is within bounds
        if (idx + 2 < size) {
            // Get the color data for the current group
            uint8_t r = data[idx];
            uint8_t g = data[idx + 1];
            uint8_t b = data[idx + 2];
            
            // Apply the intensity scaling if needed
            if (globalIntensity != 255) {
                r = r * globalIntensity / 255;
                g = g * globalIntensity / 255;
                b = b * globalIntensity / 255;
            }
            
            // Update all LEDs in the current group with the same color data
            for (size_t pixel = 0; pixel < globalgroup1; ++pixel) {
                size_t ledIndex = startPixel + group * globalgroup1 + pixel;
                
                if (ledIndex < startPixel + pixelCount) {
                    if (globalOrder == "RGB") {
                        leds[ledIndex].setRGB(r, g, b);
                    } else if (globalOrder == "GRB") {
                        leds[ledIndex].setRGB(g, r, b);
                    } else if (globalOrder == "BRG") {
                        leds[ledIndex].setRGB(b, r, g);
                    } else if (globalOrder == "GBR") {
                        leds[ledIndex].setRGB(g, b, r);
                    } else {
                        leds[ledIndex].setRGB(r, g, b);
                    }
                }
            }
        }
    }
}

artnet.subscribeArtDmx([&](const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {

    // Check which universe is being used
    if (metadata.universe == 0) {
        //Serial.println("Handling Universe 0");
        handleUniverse(data, size, 0, 170);  // First segment of pixels
    } else if (metadata.universe == 1) {
        //Serial.println("Handling Universe 1");
        handleUniverse(data, size, 170, 70); // Second segment of pixels
    }
    });
    ```

hideakitai, 2024-10-22:
@adeloscope @dantheman-22 Sorry for not getting back to you sooner. Please use `artnet.send(ip, universe, data_ptr, size);` multiple times as you want. It works correctly if your board has enough memory to store the pixels.

hideakitai, 2024-11-07:
If you still need help, please reopen this issue.

closed 2024-11-07 by hideakitai