hideakitai/ArtNet #128: Choosing between WiFi and Ethernet during runtime

baender, 2024-12-19:
This issue is similar to #85 but I was not able to solve it the way it was suggested.

Situation:

The user shall be able to choose between Art-Net via WiFi or Ethernet during runtime.

Initially I thought, I could pass the interface of choice to a generalized Artnet class (e.g an `artnet.setInterface()` method) . I realized, that this is not possible. Therefore, I tried to have one instance of each class using only one of the two.

When including `ArtnetWiFi.h` AND `ArtnetEther.h` I get a compiler error, when parsing Artnet (ethernet). That is not surprising since the library documentation explicitly says NOT to include both.

Here is my minimal example:

```cpp
#include <ArtnetWiFi.h>
#include <ArtnetEther.h>

ArtnetWiFiReceiver artnetWiFi;
ArtnetReceiver artnetEthernet;

bool useWiFi;

void receiveArtNetWiFiCallback(const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
  // do nothing
}

void receiveArtNetEthernetCallback(const uint8_t *data, uint16_t size, const ArtDmxMetadata &metadata, const ArtNetRemoteInfo &remote) {
  // do nothing
}

void setup() {
  useWiFi = false;  // defined during runtime

  if (useWiFi) {
    WiFi.begin("ssid", "password");
    artnetWiFi.begin();
    artnetWiFi.subscribeArtDmxUniverse(0, receiveArtNetWiFiCallback);
  } else {
    byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
    Ethernet.begin(mac);
    artnetEthernet.begin();
    artnetEthernet.subscribeArtDmxUniverse(0, receiveArtNetEthernetCallback);
  }
}

void loop() {
  if (useWiFi) {
    artnetWiFi.parse();
  } else {
    artnetEthernet.parse();  // Triggers the compiler error
  }
}
```

These are the compiler errors

```
.pio/libdeps/arduino_nano_esp32/ArtNet/Artnet/Receiver.h:378:25: error: no matching function for call to 'art_net::Receiver_<EthernetUDP>::localIP()'

.pio/libdeps/arduino_nano_esp32/ArtNet/Artnet/Receiver.h:380:9: error: no matching function for call to 'art_net::Receiver_<EthernetUDP>::macAddress(uint8_t [6])'
```

I wonder if there is any way I can support both interfaces on runtime.

Information to my setup:
Board: Arduino Nano ESP32
ArtNet: 0.8.0
Ethernet: 2.0.2

hideakitai, 2024-12-22:
Making multiple interfaces available would be possible, but this is not currently planned.

Out of curiosity, in what situations would both Ethernet and WiFi interfaces be needed?

svinux, 2024-12-22:
The one could be the Backup of the other ohne if for example wired would fail you could use WiFi till you fix wired again

hideakitai, 2024-12-22:
Is it difficult to switch Ethernet/WiFi in the preprocessor (e.g., #ifdef) and rewrite the rewriteram again? Do you want to make it so that it is impossible to rewrite, and if you start the program while pressing some physical button, it will switch to Ethernet/WiFi?

baender, 2024-12-22:
> Is it difficult to switch Ethernet/WiFi in the preprocessor (e.g., #ifdef) and rewrite the program again? Do you want to make it so that it is impossible to rewrite, and if you start the program while pressing some physical button, it will switch to Ethernet/WiFi?

Hi hideakitai,

I am working on an Artnet-device I want to give to a user. The user should be able to use the device without setting up an environment and programming the device. Instead they should be able to use a graphical user interface (display and rotary encoder or web interface) to configure the device.

The user might not have WiFi available or is worried about interference and would rather choose cable (Ethernet).

On the other hand, the device might need to work without cable (e.g. battery powered) and only WiFi is an option.

I want the user to be able to choose between the two Interfaces depending on their needs. But again, without the need of programming the device only via a user interface (during runtime).

hideakitai, 2024-12-23:
Okay, I see several requests to use both Wi-Fi and Ethernet. I don't know when I can support this library, as I am working on it in my spare time, but I will look into improving the ability to use multiple backends simultaneously. Of course, pull requests are welcome.

kilrah, 2025-02-23:
+1 for runtime selection.
For a "production" setup I'd of course always use ethernet, but it'd be convenient to be able to switch the same device to wifi for config/testing without having to go dig out NICs/switch/cables.

thirstyice, 2025-06-22:
Here's the way I've allowed for runtime switching between Wifi/Ethernet:

Create an `ArtnetProtocol` class defining pure virtual functions for interacting with artnet (`ArtnetProtocol.h`)
Create an `ArtnetWifiProtocol` class inheriting from and overriding  `ArtnetProtocol` (`ArtnetWifiProtocol.h`)
Create an `ArtnetEthernetProtocol` class also inheriting from and overriding `ArtnetProtocol` (`ArtnetEthernetProtocol.h`)
 - Note that none of these header files should include the Artnet library headers! They only contain their respective class declarations!

In `ArtnetWifiProtocol.cpp`, include `<ArtnetWiFi.h>` and use it to implement `ArtnetProtocolWifi`
In `ArtnetEthernetProtocol.cpp`, include`<ArtnetETH.h>` (or whichever flavour of ethernet is applicable) and use it to implement `ArtnetProtocolEthernet`

In the main application:
Define an instance of `ArtnetWifiProtocol` for interacting w/ artnet over wifi
Define an instance of `ArtnetEthernetProtocol` for interacting w/ artnet over ethernet
Define an `ArtnetProtocol*` for interacting w/ whichever implementation you choose at runtime (if `ArtnetProtocol* == &ArtnetWifiProtocol` use Wifi, if `ArtnetProtocol* == &ArtnetEthernetProtocol` use Ethernet)

Huzzah!

thirstyice, 2025-06-25:
I feel like adding an interface on top of the existing library functions probably isn't the best solution for inclusion in the library itself as it's basically an entirely separate entity, but it may be possible to re-write the library to use a similar structure. I'm not sure how much that would affect the current API though.
I'll look into it when I have time.

hideakitai, 2025-10-17:
I attempted to build send_receive_eth.ino on the ESP32-C3 using PlatformIO. It seems to link successfully, but I wonder why.

```
$ pio ci --board esp32-c3-devkitm-1 --project-option="framework=arduino" --project-option="platform=https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip" --project-option="lib_deps=https://github.com/hideakitai/ArtNet.git"  send_receive_eth.ino
...
Archiving .pio/build/esp32-c3-devkitm-1/libFrameworkArduino.a
Indexing .pio/build/esp32-c3-devkitm-1/libFrameworkArduino.a
Linking .pio/build/esp32-c3-devkitm-1/firmware.elf
Retrieving maximum program size .pio/build/esp32-c3-devkitm-1/firmware.elf
Checking size .pio/build/esp32-c3-devkitm-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.3% (used 40148 bytes from 327680 bytes)
Flash: [=======   ]  73.1% (used 958110 bytes from 1310720 bytes)
Building .pio/build/esp32-c3-devkitm-1/firmware.bin
esptool.py v4.8.5
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
===================================== [SUCCESS] Took 10.93 seconds =====================================
```

kilrah, 2025-10-17:
Sorry, found the issue, I had `#include <ArtnetETH.h>` getting into another source file than the main, removing that builds fine.