hideakitai/ArtNet #129: Delayed response if more than 2x Artnet universes are received

mediaMicha, 2024-12-26:
Hello,

first of all, thank you very much for providing this great library. 
I use ESP-WROOM-32 and ESP32-S3 with the Ethernet module W5500 with the same response. As long as I only send one or two Artnet universes from the Grand MA3, the ESP receives / reacts to every change immediately. And it doesn't matter whether these are distributed in one or more subnets. With more than 2, i.e. 5 or so, the interface receives / reacts very bad.
Can you please see if you can get this fixed? If there is anything I can do to help, please let me know.

And then I have a second question, when should the “callback” function be used?

hideakitai, 2025-02-02:
> Can you please see if you can get this fixed? If there is anything I can do to help, please let me know.

Could you confirm which of your networks or ESP32 has been delayed? (I think your network causes it) You can see the packets using WireShark or if you set the debug logger as follows:

```cpp
artnet.setLogger(&Serial);
```

You can see the following log when you received a packet:

```
Packet received: size = ....
```

> And then I have a second question, when should the “callback” function be used?

The callback function should be used to do something when you receive (and parse successfully) an ArtNet packet.

hideakitai, 2025-02-03:
If you still need help, please reopen this issue

closed 2025-02-03 by hideakitai

mediaMicha, 2025-02-06:
Hello, I'm a bit busy at the moment. I'll make sure I get around to testing soon.

mediaMicha, 2025-09-01:
We can use Broadcast without the universe of this artnet node and send the data of this universe as unicast, so its work fine. Only this artnet node is in broadcast included, we have the delay.
I hope this information helps to find the issue. If we can support you, please let me know.
Thank you for your support.